linux-nvme/nvme-cli #203: Unable to find property Get command and property Set command in nvme cli

ksubbiahx, 2017-07-14:
Hi,

Is there any option to use the property get and property set commands in nvme-cli 1.3 to read and set the value of a property.

keithbusch, 2017-07-14:
Are you referring to fabrics properties?

ksubbiahx, 2017-07-17:
Property of a controller that can be either a Dword, or Qword attribute. The host shall access properties in their native width with an offset that is at the beginning of the property. 
It is supported as per spec. For example, offset 14h for getting controller configuration property.

keithbusch, 2017-07-17:
Okay, you are referring to fabrics controller properties. With PCI, these are called "registers" and are accessed with memory mapped IO (if the kernel supports it) using the "show-regs" command. For fabrics, we need to add support for the property get, command and that is not done as yet. I don't think there's a good use case for making it easy to set a property, though.

tylarb, 2019-10-01:
@keithbusch  I don't see a "set-regs" command - would it be possible to implement one? I'm trying to set a bad CSTS to see if we could reproduce a hardware failure.

keithbusch, 2019-10-08:
I really don't think we want to make it easy to do such a thing. I can recommend Bill Farrow's "pcimem" utility though, located at https://github.com/billfarrow/pcimem