linux-nvme/nvme-cli #223: Nvme-write command

SBoldenn, 2017-08-23:
Hello,

Can someone please update the nvme-write documentation with an example ?
Specifically an example involving 'PRINFO' field.

I have formatted my Intel drive to support Protection Type 1. But for some reason when I issue a write command with PRINFO=0x1 - CLI stops responding and write result is not shown

keithbusch, 2017-08-25:
Sorry, I'm still way behind on getting all the examples added to the documentation.

For the intel drives, these only support interleaved metadata, so you'll need to specify the data + metadata transfer sizes together. For example, if using LBAF 1 (512B + 8B Metadata), we have to use 520 byte transfers:

`echo "hello world" | nvme write /dev/nvme0n1 --data-size=520 --block-count=0
`
The important part here is that we told nvme-cli to use a 520 byte buffer, zero padded beyond the "hello world" part.

The tool will let you specify any value here, but if your data size and block count don't correctly account for the actual transfer, you risk data and/or memory corruption.

Specifically on your question, just appending "`--prinfo=1`" will set the prinfo field of the command as you are requesting, which should have the reference tag compared. I don't see any "hang" when I test this, and the results are near immediate. I do not, however, observe the controller actually verifying the reference tag, though (might just be the firmware I'm running).

keithbusch, 2017-08-25:
```
# echo "hello world" | nvme write /dev/nvme0n1 --data-size=520 --prinfo=1
write: Success
# nvme read /dev/nvme0n1 --data-size=520 --prinfo=1
hello world
read: Success
```

SBoldenn, 2017-08-28:
Hello Keith

Thanks a bunch for the reply. I tried to issue a "write" command as per your suggestion. But looks like the issue is with Intel drive (model no. DC P3700).  Not sure if the firmware on the drive is out of date. 

I have few other drives from other vendors that either complete successfully or return an error (APP Tag error, Guard Error, Ref Tag error) but do not _**hang**_. 

Thanks again for the pointers

keithbusch, 2017-08-29:
Is the completion successful when you do not add the prinfo field?

After setting a metadata format, have you either reset or power cycled the controller? The spec does not require it, but the P3x00 family of drives all require a reset after changing the block size.

closed 2022-01-19 by igaw