linux-nvme/nvme-cli #225: nvme list in json and normal formats show different fields

baruch, 2017-09-24:
I was hoping to use the output of `nvme list` to get the namespace of the device, this is shown in the normal format but when I tried to use the json format this information is not printed in the json format which makes it needlessly harder to get the data in a script.

Please make it so that all the data in one format is available in the other format as well.

keithbusch, 2017-09-25:
That seems reasonable to me.

michael-db, 2020-12-22:
The particular consistency problem mentioned appears to have been fixed, but there is a related one: the output of "nvme list -o json" includes DevicePath but the output of "nvme list -o json -v", which should be a superset of the former, does not. "NameSpace" in the former is what is called NSID in the latter, i.e., a number, and "Namespace" in the latter is the former's "DevicePath" but without the "/dev/". Is that going to be fixed? And if it is, how many scripts will it break?
```
ubuntu$ nvme --version
nvme version 1.13.21.g55adf
```
```
  "Devices" : [
    {
      "NameSpace" : 1,
      "DevicePath" : "/dev/nvme1n1",
```
```
      "Namespaces" : [
        {
          "NameSpace" : "nvme1n1",
          "NSID" : 1,
```

igaw, 2022-01-19:
No progess in this bug report for a long time. If you think it's still relevant please reopen.

closed 2022-01-19 by igaw