linux-nvme/nvme-cli #226: how to uninstall nvme 1.3

visheshvijay, 2017-10-03:
Hi,

I would like to know how to uninstall the nvme 1.3 on Rhel_7.3
I am trying to install the nvme 1.4 but I am seeing this error 

[root@aic-node-a nvme-cli-1.4]# make
cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -DNVME_VERSION='"1.4"' -c argconfig.c
make: cc: Command not found
make: *** [argconfig.o] Error 127
[root@aic-node-a nvme-cli-1.4]#

Thanks,
Vishesh

visheshvijay, 2017-10-03:
Installation is successful after updating the gcc but let us know the process of uninstalling the nvme_cli

keithbusch, 2017-10-03:
the 'make install' should replace all the old artifacts. We don't have an uninstall target in the makefile.

igaw, 2022-01-19:
No progess in this bug report for a long time. If you think it's still relevant please reopen.

closed 2022-01-19 by igaw