linux-nvme/nvme-cli #227: Failed to create soft link on /sys

mfinkelstine, 2017-10-16:
Hi ,

I used nvme-cli on my CentOS 7.3 with kernel 4.12.3-1.el7.centos.x86_64.
And I get an error in the end of the procedure to create testnqn

here is the command output :
root@testhost:~ $ nvmetcli
/ports/3/subsystems> create testnqn
Could not symlink testnqn in configFS: [Errno 22] Invalid argument

this kernel have been modified and compiled to use NVMe 
any idea why?

Thanks

keithbusch, 2017-10-16:
Is your kernel configured to support nvme target? If so, is configfs mounted?

ksangeek, 2018-02-23:
I am facing the same issue.
How do i check if kernel is configured to support nvme target ?
I have the kernel modules loaded. Is that not sufficient :
```
# lsmod | grep nvme
nvmet_fc               23807  0
nvmet                  55161  11 nvmet_fc
nvme                   32028  0
nvme_core              48426  1 nvme
```

In my case `configfs` is mounted.

keithbusch, 2018-02-23:
Looks like you've nvmet, so that should mean nvme target is configured.

To be honest, though, I'm not super familiar with nvmetcli. That project is not related to this one.

ksangeek, 2018-02-24:
Any tips on how to get to the root cause of this issue ?