linux-nvme/nvme-cli #239: flush command should send to controller or namespace?

rolandqi, 2017-12-13:
I have a nvme device that have /dev/nvme0 and /dev/nvme0n1.
I can send 1: nvme flush /dev/nvme0n1 -n 1   
                 2: nvme flush /dev/nvme0 -n 1
Both command can pass but if I use 2nd cmd, dmesg would show: nvme nvme0: using deprecated NVME_IOCTL_IO_CMD ioctl on the char device!  
does this means sending to nvme0 is a wrong ?

keithbusch, 2017-12-13:
Yes, sending IO commands to the controller handle is deprecated. In fact, the Linux kernel today will allow it with this warning only if you have a single namespace. If you've multiple namespace, the ioctl will print a different warning and return an error. The reason for this is that allowing generic IO through this ioctl breaks isolation. You may have multiple namespaces assigned to different containers, and you wouldn't want one container to be able to snoop data from another through some universal access mechanism.

rolandqi, 2017-12-14:
Got it! thanks a lot! It really helpful.

closed 2017-12-14 by keithbusch

narasimhan-v, 2018-01-05:
@keithbusch I have single namespace, and I see this command sent to controller passing and namespace failing. Why is that so ?

```
# nvme flush /dev/nvme0 -n 1
[  709.456888] nvme nvme0: using deprecated NVME_IOCTL_IO_CMD ioctl on the char device!
NVMe Flush: success

# nvme flush /dev/nvme0n1
NVME IO command error:INVALID_NS(b)
# nvme version
nvme version 1.3
```

keithbusch, 2018-01-05:
You can override the default with '-n 1', but the controller is supposed to accept a flush command with 'all NSIDs'. FFFFFFFFh. It sounds like your controller is out of spec compliance here.

narasimhan-v, 2018-01-06:
Thanks @keithbusch,  will check with the vendor.