linux-nvme/nvme-cli #241: nvme cli crashes on reading 65536 blocks of 4K ns

visheshvijay, 2017-12-21:
Hi Keith ,

My environment is 
RHEL_7.3
nvme cli 1.4
Namespace - 4K

So below read command would read 256MB of data at once but I am observing crash

[root@jaguar54 ~]# nvme read /dev/nvme0n1 -s 0 -c 65535 -z 268435456 -d data
read: Success
*** Error in `-bash': free(): invalid pointer: 0x00000000011e23c0 ***
======= Backtrace: =========
/lib64/libc.so.6(+0x7c503)[0x7f37a3691503]
-bash(dispose_word+0x11)[0x42e261]
-bash(dispose_words+0x86)[0x42e436]
-bash(dispose_command+0x125)[0x42e615]
-bash(reader_loop+0xab)[0x41e19b]
-bash(main+0xc4e)[0x41c8ee]
/lib64/libc.so.6(__libc_start_main+0xf5)[0x7f37a3636b35]
-bash[0x41d38d]
======= Memory map: ========
00400000-004dd000 r-xp 00000000 08:01 5506010                            /usr/bin/bash
006dc000-006dd000 r--p 000dc000 08:01 5506010                            /usr/bin/bash
006dd000-006e6000 rw-p 000dd000 08:01 5506010                            /usr/bin/bash
006e6000-006ec000 rw-p 00000000 00:00 0
010ac000-01253000 rw-p 00000000 00:00 0                                  [heap]
7f3798000000-7f3798021000 rw-p 00000000 00:00 0
7f3798021000-7f379c000000 ---p 00000000 00:00 0
7f379ccc3000-7f379ccd8000 r-xp 00000000 08:01 5505043                    /usr/lib64/libgcc_s-4.8.5-20150702.so.1
7f379ccd8000-7f379ced7000 ---p 00015000 08:01 5505043                    /usr/lib64/libgcc_s-4.8.5-20150702.so.1
7f379ced7000-7f379ced8000 r--p 00014000 08:01 5505043                    /usr/lib64/libgcc_s-4.8.5-20150702.so.1
7f379ced8000-7f379ced9000 rw-p 00015000 08:01 5505043                    /usr/lib64/libgcc_s-4.8.5-20150702.so.1
7f379ced9000-7f37a3402000 r--p 00000000 08:01 5899735                    /usr/lib/locale/locale-archive
7f37a3402000-7f37a340e000 r-xp 00000000 08:01 5505957                    /usr/lib64/libnss_files-2.17.so
7f37a340e000-7f37a360d000 ---p 0000c000 08:01 5505957                    /usr/lib64/libnss_files-2.17.so
7f37a360d000-7f37a360e000 r--p 0000b000 08:01 5505957                    /usr/lib64/libnss_files-2.17.so
7f37a360e000-7f37a360f000 rw-p 0000c000 08:01 5505957                    /usr/lib64/libnss_files-2.17.so
7f37a360f000-7f37a3615000 rw-p 00000000 00:00 0
7f37a3615000-7f37a37cb000 r-xp 00000000 08:01 5505939                    /usr/lib64/libc-2.17.so
7f37a37cb000-7f37a39cb000 ---p 001b6000 08:01 5505939                    /usr/lib64/libc-2.17.so
7f37a39cb000-7f37a39cf000 r--p 001b6000 08:01 5505939                    /usr/lib64/libc-2.17.so
7f37a39cf000-7f37a39d1000 rw-p 001ba000 08:01 5505939                    /usr/lib64/libc-2.17.so
7f37a39d1000-7f37a39d6000 rw-p 00000000 00:00 0
7f37a39d6000-7f37a39d8000 r-xp 00000000 08:01 5505945                    /usr/lib64/libdl-2.17.so
7f37a39d8000-7f37a3bd8000 ---p 00002000 08:01 5505945                    /usr/lib64/libdl-2.17.so
7f37a3bd8000-7f37a3bd9000 r--p 00002000 08:01 5505945                    /usr/lib64/libdl-2.17.so
7f37a3bd9000-7f37a3bda000 rw-p 00003000 08:01 5505945                    /usr/lib64/libdl-2.17.so
7f37a3bda000-7f37a3bff000 r-xp 00000000 08:01 5506008                    /usr/lib64/libtinfo.so.5.9
7f37a3bff000-7f37a3dff000 ---p 00025000 08:01 5506008                    /usr/lib64/libtinfo.so.5.9
7f37a3dff000-7f37a3e03000 r--p 00025000 08:01 5506008                    /usr/lib64/libtinfo.so.5.9
7f37a3e03000-7f37a3e04000 rw-p 00029000 08:01 5506008                    /usr/lib64/libtinfo.so.5.9
7f37a3e04000-7f37a3e24000 r-xp 00000000 08:01 5505932                    /usr/lib64/ld-2.17.so
7f37a400b000-7f37a400e000 rw-p 00000000 00:00 0
7f37a4018000-7f37a401a000 rw-p 00000000 00:00 0
7f37a401a000-7f37a4021000 r--s 00000000 08:01 5899734                    /usr/lib64/gconv/gconv-modules.cache
7f37a4021000-7f37a4023000 rw-p 00000000 00:00 0
7f37a4023000-7f37a4024000 r--p 0001f000 08:01 5505932                    /usr/lib64/ld-2.17.so
7f37a4024000-7f37a4025000 rw-p 00020000 08:01 5505932                    /usr/lib64/ld-2.17.so
7f37a4025000-7f37a4026000 rw-p 00000000 00:00 0
7ffe028cc000-7ffe028ed000 rw-p 00000000 00:00 0                          [stack]
7ffe028fe000-7ffe02900000 r-xp 00000000 00:00 0                          [vdso]
ffffffffff600000-ffffffffff601000 r-xp 00000000 00:00 0                  [vsyscall]

keithbusch, 2017-12-21:
Hm, it sounds like your drive is overrunning that buffer. Is 256MB the threshold where the bug occurs?

visheshvijay, 2017-12-29:
Hi Keith,

I have narrowed down it when count is greater than 32 then cli crashes and kernel also crashes
Can you please confirm whether its a cli bug  or drive bug

[root@jaguar54 ~]# nvme read /dev/nvme0n1 -s 0 -c 31 -z 409600 -d data
read: Success
[root@jaguar54 ~]# nvme read /dev/nvme0n1 -s 0 -c 32 -z 409600 -d data
read: Success
-bash: wait_for: No record of process 3794
*** Error in `-bash': free(): invalid pointer: 0x0000000000cedc90 ***
======= Backtrace: =========
/lib64/libc.so.6(+0x7c503)[0x7f81b1cf0503]
-bash(dispose_words+0x86)[0x42e436]
-bash(dispose_command+0x125)[0x42e615]
-bash(reader_loop+0xab)[0x41e19b]
-bash(main+0xc4e)[0x41c8ee]
/lib64/libc.so.6(__libc_start_main+0xf5)[0x7f81b1c95b35]
-bash[0x41d38d]
======= Memory map: ========
00400000-004dd000 r-xp 00000000 08:01 5506010                            /usr/bin/bash
006dc000-006dd000 r--p 000dc000 08:01 5506010                            /usr/bin/bash
006dd000-006e6000 rw-p 000dd000 08:01 5506010                            /usr/bin/bash
006e6000-006ec000 rw-p 00000000 00:00 0
00ce3000-00e6f000 rw-p 00000000 00:00 0                                  [heap]
7f81a4000000-7f81a4021000 rw-p 00000000 00:00 0
7f81a4021000-7f81a8000000 ---p 00000000 00:00 0
7f81ab322000-7f81ab337000 r-xp 00000000 08:01 5505043                    /usr/lib64/libgcc_s-4.8.5-20150702.so.1
7f81ab337000-7f81ab536000 ---p 00015000 08:01 5505043                    /usr/lib64/libgcc_s-4.8.5-20150702.so.1
7f81ab536000-7f81ab537000 r--p 00014000 08:01 5505043                    /usr/lib64/libgcc_s-4.8.5-20150702.so.1
7f81ab537000-7f81ab538000 rw-p 00015000 08:01 5505043                    /usr/lib64/libgcc_s-4.8.5-20150702.so.1
7f81ab538000-7f81b1a61000 r--p 00000000 08:01 5899735                    /usr/lib/locale/locale-archive
7f81b1a61000-7f81b1a6d000 r-xp 00000000 08:01 5505957                    /usr/lib64/libnss_files-2.17.so
7f81b1a6d000-7f81b1c6c000 ---p 0000c000 08:01 5505957                    /usr/lib64/libnss_files-2.17.so
7f81b1c6c000-7f81b1c6d000 r--p 0000b000 08:01 5505957                    /usr/lib64/libnss_files-2.17.so
7f81b1c6d000-7f81b1c6e000 rw-p 0000c000 08:01 5505957                    /usr/lib64/libnss_files-2.17.so
7f81b1c6e000-7f81b1c74000 rw-p 00000000 00:00 0
7f81b1c74000-7f81b1e2a000 r-xp 00000000 08:01 5505939                    /usr/lib64/libc-2.17.so
7f81b1e2a000-7f81b202a000 ---p 001b6000 08:01 5505939                    /usr/lib64/libc-2.17.so
7f81b202a000-7f81b202e000 r--p 001b6000 08:01 5505939                    /usr/lib64/libc-2.17.so
7f81b202e000-7f81b2030000 rw-p 001ba000 08:01 5505939                    /usr/lib64/libc-2.17.so
7f81b2030000-7f81b2035000 rw-p 00000000 00:00 0
7f81b2035000-7f81b2037000 r-xp 00000000 08:01 5505945                    /usr/lib64/libdl-2.17.so
7f81b2037000-7f81b2237000 ---p 00002000 08:01 5505945                    /usr/lib64/libdl-2.17.so
7f81b2237000-7f81b2238000 r--p 00002000 08:01 5505945                    /usr/lib64/libdl-2.17.so
7f81b2238000-7f81b2239000 rw-p 00003000 08:01 5505945                    /usr/lib64/libdl-2.17.so
7f81b2239000-7f81b225e000 r-xp 00000000 08:01 5506008                    /usr/lib64/libtinfo.so.5.9
7f81b225e000-7f81b245e000 ---p 00025000 08:01 5506008                    /usr/lib64/libtinfo.so.5.9
7f81b245e000-7f81b2462000 r--p 00025000 08:01 5506008                    /usr/lib64/libtinfo.so.5.9
7f81b2462000-7f81b2463000 rw-p 00029000 08:01 5506008                    /usr/lib64/libtinfo.so.5.9
7f81b2463000-7f81b2483000 r-xp 00000000 08:01 5505932                    /usr/lib64/ld-2.17.so
7f81b266a000-7f81b266d000 rw-p 00000000 00:00 0
7f81b2677000-7f81b2679000 rw-p 00000000 00:00 0
7f81b2679000-7f81b2680000 r--s 00000000 08:01 5899734                    /usr/lib64/gconv/gconv-modules.cache
7f81b2680000-7f81b2682000 rw-p 00000000 00:00 0
7f81b2682000-7f81b2683000 r--p 0001f000 08:01 5505932                    /usr/lib64/ld-2.17.so
7f81b2683000-7f81b2684000 rw-p 00020000 08:01 5505932                    /usr/lib64/ld-2.17.so
7f81b2684000-7f81b2685000 rw-p 00000000 00:00 0
7ffdc097a000-7ffdc099b000 rw-p 00000000 00:00 0                          [stack]
7ffdc09ec000-7ffdc09ee000 r-xp 00000000 00:00 0                          [vdso]
ffffffffff600000-ffffffffff601000 r-xp 00000000 00:00 0                  [vsyscall]

keithbusch, 2017-12-29:
Could you attach the output from "nvme id-ctrl /dev/nvme0 | grep mdts"?

visheshvijay, 2018-01-30:
Keith,

Anyways I am not seeing this issue anymore. If you want you can close this issue

-Vishesh

closed 2018-01-30 by keithbusch