linux-nvme/nvme-cli #244: Build Failure on Centos 6.8

sppavilion, 2017-12-29:
Version Information
================================
[root@localhost nvme-cli]# cat /etc/centos-release 
CentOS release 6.8 (Final)
[root@localhost nvme-cli]# uname -r
2.6.32-642.15.1.el6.x86_64
[root@localhost nvme-cli]# 

Steps followed
================================
Downloaded nvme-cli 1.4 and tried make command.
---------------------------------------------------
[root@localhost nvme-cli]# make
NVME_VERSION = 1.4
cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -DNVME_VERSION='"1.4"' -c argconfig.c
cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -DNVME_VERSION='"1.4"' -c suffix.c
cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -DNVME_VERSION='"1.4"' -c parser.c
cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -DNVME_VERSION='"1.4"' -c nvme-print.c
cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -DNVME_VERSION='"1.4"' -c nvme-ioctl.c
cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -DNVME_VERSION='"1.4"' -c nvme-lightnvm.c
cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -DNVME_VERSION='"1.4"' -c fabrics.c
cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -DNVME_VERSION='"1.4"' -c json.c
cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -DNVME_VERSION='"1.4"' -c plugin.c
cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -DNVME_VERSION='"1.4"' -c intel-nvme.c
cc1: warnings being treated as errors
intel-nvme.c: In function ‘get_internal_log’:
intel-nvme.c:782: error: dereferencing pointer ‘ehdr’ does break strict-aliasing rules
intel-nvme.c:785: error: dereferencing pointer ‘ehdr’ does break strict-aliasing rules
intel-nvme.c:741: note: initialized from here
intel-nvme.c:643: error: ‘ad’ may be used uninitialized in this function
intel-nvme.c:645: error: ‘ehdr’ may be used uninitialized in this function
make: *** [intel-nvme.o] Error 1

Thanks,
Shubham

keithbusch, 2018-01-02:
Are you using the tagged version of 1.4 or the latest 1.4 after that tag? This should be fixed with commit b3b4cb86a67c596fd18d9c7ac719b630641c2ed0.

sppavilion, 2018-01-03:
Keith,

I am using release version of nvme-cli 1.4 from https://github.com/linux-nvme/nvme-cli/archive/v1.4.zip . I am not facing compile issue with latest one.

But I want to use only released version. Should I continue with nvme-cli 1.3 for Centos 6.8/6.9

Thanks,

Shubham

keithbusch, 2018-01-03:
Thanks for confirming. I'll just tag a new release, v1.5. It's been long enough, anyway.

sppavilion, 2018-01-03:
Thanks a lot, Keith !!!

keithbusch, 2018-01-03:
v1.5 tagged and pushed; release available.

closed 2018-01-03 by keithbusch