linux-nvme/nvme-cli #245: nand and host_bytes_written field not visible in smart-log-add command

dhkoo, 2018-01-01:
[kau@pm004 nvme-cli-1.0]$ sudo ./nvme smart-log-add /dev/nvme0
Additional Smart Log for NVME device:nvme0 namespace-id:ffffffff
key                               normalized raw
program_fail_count              : 100%       0
erase_fail_count                : 100%       0
wear_leveling                   : 100%       min: 1, max: 6, avg: 3
end_to_end_error_detection_count: 100%       0
crc_error_count                 : 100%       85385
timed_workload_media_wear       : 100%       0.030%
timed_workload_host_reads       : 100%       0%
timed_workload_timer            : 100%       661523 min
thermal_throttle_status         : 100%       0%, cnt: 0
retry_buffer_overflow_count     : 100%       0
pll_lock_loss_count             : 100%       0
**nand_bytes_written              :   0%       sectors: 0
host_bytes_written              :   0%       sectors: 0**

I think nand and host_bytes_written field is not working.
It works well on other machines.

What is the cause? And what is the solution?

keithbusch, 2018-01-02:
What is the model of the device you're running this against?

tim-oleksii, 2018-01-11:
program_fail_count : 100% 0
erase_fail_count : 100% 0
Did you test nvme-cli against the same device on other PC, or just other ssd?

keithbusch, 2018-01-11:
Reporter needs to inform what device returned this data. Not all devices support this non-standard log page.

igaw, 2022-01-19:
No progess in this bug report for a long time. If you think it's still relevant please reopen.

closed 2022-01-19 by igaw