linux-nvme/nvme-cli #252: Garbage build files

tim-oleksii, 2018-01-08:
There are huge amount of build artifacts in the documentation:
- *.1
- *.html

Does gitignore need to be updated? Is it intended to keep them?

keithbusch, 2018-01-08:
This is as intended. I don't necessarily want to carry these auto-generated artifacts, but if I make it a requirement that you have asciidoc and xmlto just to build tooling, I'll just get to repeatedly explain this dependency through many broken builds issues.

tim-oleksii, 2018-01-08:
I'm sorry, it's quite unclear to me, could you please give more insights?

Here is my understanding

The tool could be built by following:
- make (tool, no documentation)
- make nvme (tool, no documentation)
- make all (tool, documentation)

Documentation only:
- make doc

only make {all|doc} require to have asciidoc and xmlto installed. If ones uses make all, then asciidoc and xmlto are already installed. If someone are up to setting build process via make all, then clear message regarding dependencies is seen.

How is it going to break build process?

Thanks!

keithbusch, 2018-01-08:
Here's the experiment to run:

1. Start with a clean clone of the repository
2. Remove all the .1 and .html files in Documentation
3. Run 'make install'

tim-oleksii, 2018-01-08:
Ah, I see, thank you. Let's close the issue.

closed 2018-01-08 by tim-oleksii