linux-nvme/nvme-cli #260: intel smart-log-add help differs from the man page

tim-oleksii, 2018-01-12:
Man page differs from help message
> [tim-oleksii@archlinux nvme-cli]$ ./nvme intel help smart-log-add

> SYNOPSIS
>        nvme intel smart-log-add <device> [--namespace-id=<nsid> | -n <nsid>]
>                                [--raw-binary | -b]
>                                [--output-format=<fmt> | -o <fmt>]

However, issuing wrong command prints other thing

> [tim-oleksii@archlinux nvme-cli]$ ./nvme intel smart-log-add blablabla
> blablabla: No such file or directory
> Usage: nvme intel smart-log-add <device> [OPTIONS]
> 
> Get Intel vendor specific additional smart log (optionally, for the specified
> namespace), and show it.
> 
> Options:
>   [  --namespace-id=<NUM>, -n <NUM> ]   --- (optional) desired namespace
>   [  --raw-binary, -b ]                 --- dump output in binary format
>   [  --json, -j ]                       --- Dump output in json format

Version used
> [tim-oleksii@archlinux nvme-cli]$ ./nvme version
> nvme version 1.5.24.ga55d.dirty

minwooim, 2018-01-12:
Thanks for your kindly reporting.
I was trying to fix your comment in #175 with #259  which cause this issue.
I have fixed this issue with #262 pull request.
Sorry for making confusions.

Thanks,

tim-oleksii, 2018-01-15:
I see it's got its way into the main branch, let's close the ticket. Thx for a fix!

closed 2018-01-15 by tim-oleksii