linux-nvme/nvme-cli #265: intel smart-log-add doesn't utilize identify

tim-oleksii, 2018-01-12:
I wonder why intel doesn't call id_ctrl to get VID or MN and check it against expected values?
I'm able to get garbage additional SMARTs from Hynix device.

For instance,

> # ./nvme intel smart-log-add /dev/nvme0n1 
> Additional Smart Log for NVME device:nvme0n1 namespace-id:ffffffff
> key                               normalized raw
> program_fail_count              :   0%       0
> erase_fail_count                :   0%       45136
> wear_leveling                   :   0%       min: 0, max: 0, avg: 0
> end_to_end_error_detection_count:   0%       0
> crc_error_count                 :   0%       0
> timed_workload_media_wear       :   0%       0.000%
> timed_workload_host_reads       :   0%       429496729600%
> timed_workload_timer            :   0%       100 min
> thermal_throttle_status         :   0%       0%, cnt: 0
> retry_buffer_overflow_count     :   0%       131072
> pll_lock_loss_count             :   0%       0
> nand_bytes_written              :   0%       sectors: 0
> host_bytes_written              :   0%       sectors: 0
> # echo $?
> 0

> # ./nvme id-ctrl /dev/nvme0n1 -H
> NVME Identify Controller:                   
> vid     : 0x1c5c

keithbusch, 2018-01-14:
This "smart-log-add" is a vendor specific log page, and the decoding of the data to this subcommand is specific to the Intel family of controllers that implement it. It doesn't apply to any other vendor except purely by coincidence.

The tool's job is to execute the commands you asked it to. It's not the tool's job to sanity check for you, as that would put a serious burden on forward compatibility.

tim-oleksii, 2018-01-14:
I see your point, but additional SMARTs might be decoded in the wrong way which could lead to a sigsegv.

Could you please give insights on forward compatibility burden that could be stumbled upon? Checking VID via identify seems to do the trick. (basically)

I have this thoughts, because I was asked to do an extension for nvme-cli for two different customers with not identical SMART parameters and different features. Looks like I'm forced to do a check against VIP, MN, serial number.

I was just wondering why nvme-cli doesn't stick to this approach.

keithbusch, 2018-01-15:
If a new class of devices wants to implement the same log page, or perhaps Intel needs to register a second VID, the tool would refuse execute the command that root requested because it isn't on the white list proposed to be the gate keeper. That would make an older version of the tool incompatible with a new device, breaking forward compatibility, and we usually don't police root's requests anyway.

I suppose you could seg fault if some device implements the same log page identifier and transfers a larger payload than allocated, but that would only happen if the controller is broken (NUMDL is supposed to prevent that from happening). As long as the allocated buffer isn't overrun, even incorrect decoding shouldn't sigsegv.

keithbusch, 2018-01-15:
Maybe as a compromise, we could consider a whitelist check, but allow it to be overridden with an option, like `--force`.

tim-oleksii, 2018-01-15:
If someone from intel decides to disclose additional customer specific SMARTs there'll be a need to implement a whitelist. 

It's up to my street to have '--force' around.