linux-nvme/nvme-cli #269: core dumped for controller identify command with CNS 13h

aparna-ankalkote, 2018-01-16:
#nvme admin-passthru -o 6 -4 0x10013 -n 1 /dev/nvme0 -r -l 6
NVMe command result:00000000
       0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
0000: 02 00 01 00 02 00                                "......"
*** Error in `nvme': free(): invalid next size (fast): 0x0000000001463000 ***
======= Backtrace: =========
[Coredump.zip](https://github.com/linux-nvme/nvme-cli/files/1634588/Coredump.zip)

/lib64/libc.so.6(+0x7d56d)[0x7fd0c8ff856d]
nvme[0x405e77]
nvme[0x416cf0]
nvme[0x4017e1]
/lib64/libc.so.6(__libc_start_main+0xf5)[0x7fd0c8f9caf5]
nvme[0x401885]
======= Memory map: ========
00400000-00432000 r-xp 00000000 fd:00 140392724                          /usr/local/sbin/nvme
00631000-00632000 r--p 00031000 fd:00 140392724                          /usr/local/sbin/nvme
00632000-00634000 rw-p 00032000 fd:00 140392724                          /usr/local/sbin/nvme
01462000-014a4000 rw-p 00000000 00:00 0                                  [heap]
7fd0c283e000-7fd0c2853000 r-xp 00000000 fd:00 207171988                  /usr/lib64/libgcc_s-4.8.5-20150702.so.1
7fd0c2853000-7fd0c2a52000 ---p 00015000 fd:00 207171988                  /usr/lib64/libgcc_s-4.8.5-20150702.so.1
7fd0c2a52000-7fd0c2a53000 r--p 00014000 fd:00 207171988                  /usr/lib64/libgcc_s-4.8.5-20150702.so.1
7fd0c2a53000-7fd0c2a54000 rw-p 00015000 fd:00 207171988                  /usr/lib64/libgcc_s-4.8.5-20150702.so.1
7fd0c2a54000-7fd0c8f7b000 r--p 00000000 fd:00 67163654                   /usr/lib/locale/locale-archive
7fd0c8f7b000-7fd0c9131000 r-xp 00000000 fd:00 201340357                  /usr/lib64/libc-2.17.so
7fd0c9131000-7fd0c9331000 ---p 001b6000 fd:00 201340357                  /usr/lib64/libc-2.17.so
7fd0c9331000-7fd0c9335000 r--p 001b6000 fd:00 201340357                  /usr/lib64/libc-2.17.so
7fd0c9335000-7fd0c9337000 rw-p 001ba000 fd:00 201340357                  /usr/lib64/libc-2.17.so
7fd0c9337000-7fd0c933c000 rw-p 00000000 00:00 0
7fd0c933c000-7fd0c935d000 r-xp 00000000 fd:00 201340350                  /usr/lib64/ld-2.17.so
7fd0c953f000-7fd0c9542000 rw-p 00000000 00:00 0
7fd0c9559000-7fd0c955c000 rw-p 00000000 00:00 0
7fd0c955c000-7fd0c955d000 r--p 00020000 fd:00 201340350                  /usr/lib64/ld-2.17.so
7fd0c955d000-7fd0c955e000 rw-p 00021000 fd:00 201340350                  /usr/lib64/ld-2.17.so
7fd0c955e000-7fd0c955f000 rw-p 00000000 00:00 0
7fff5ff9c000-7fff5ffbd000 rw-p 00000000 00:00 0                          [stack]
7fff5fffe000-7fff60000000 r-xp 00000000 00:00 0                          [vdso]
ffffffffff600000-ffffffffff601000 r-xp 00000000 00:00 0                  [vsyscall]
Aborted (core dumped)

keithbusch, 2018-01-16:
You provide a '-l 6' param, or a transfer length of just 6 bytes. I do believe most controller list identifications will transfer 4k if that command mode is supported, so you've specified an insufficient buffer, and your results are expected in that case.

aparna-ankalkote, 2018-01-17:
I understand.
But the command was working fine with -l 6 in cli version 1.4 No core dump was observed in that.
# nvme version
nvme version 1.4
# nvme admin-passthru -o 6 -4 0x10013 -n 1 /dev/nvme0 -r -l 6
NVMe command result:00000000
       0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
0000: 02 00 01 00 02 00                                "......"

keithbusch, 2018-01-17:
1.4 would leak the memory rather than free it. The freeing is what's causing glibc to figure out you've corrupted it.

minwooim, 2018-01-17:
As Keith said, fix for memory leak until v1.4 was fixed in commit below
which is in v1.5 release commit(121dc80).

Scott Bauer (1):
      nvme-cli: Fix small mem leak. (6bc3eb)

igaw, 2022-01-19:
No progess in this bug report for a long time. If you think it's still relevant please reopen.

closed 2022-01-19 by igaw