linux-nvme/nvme-cli #272: NVME Drive not mapped to namespace

scorchpc, 2018-01-16:
I have a drive which shows up in "lspci" but does not get mapped to a namespace. 

Here is the output, let me know any questions:

root@tsgal99:~# lspci
00:00.0 Class 0600: Device 8086:0958
00:14.0 Class 0805: Device 8086:08a7 (rev 10)
00:14.1 Class 0700: Device 8086:0936 (rev 10)
00:14.2 Class 0c03: Device 8086:0939 (rev 10)
00:14.3 Class 0c03: Device 8086:0939 (rev 10)
00:14.4 Class 0c03: Device 8086:093a (rev 10)
00:14.5 Class 0700: Device 8086:0936 (rev 10)
00:14.6 Class 0200: Device 8086:0937 (rev 10)
00:14.7 Class 0200: Device 8086:0937 (rev 10)
00:15.0 Class 0c80: Device 8086:0935 (rev 10)
00:15.1 Class 0c80: Device 8086:0935 (rev 10)
00:15.2 Class 0c80: Device 8086:0934 (rev 10)
00:17.0 Class 0604: Device 8086:11c3
00:17.1 Class 0604: Device 8086:11c4
00:1f.0 Class 0601: Device 8086:095e
**01:00.0 Class 0108: Device 14a4:22f1 (rev 01)**
**root@tsgal99:~# nvme -list**
root@tsgal99:~# dmesg | grep nvme
**[   17.128610] nvme 0000:01:00.0: enabling device (0000 -> 0002)
[   17.385094] nvme 0000:01:00.0: irq 40 for MSI/MSI-X**
root@tsgal99:~#

keithbusch, 2018-01-17:
Do you have a device handle at /dev/nvme0? If so, could you run `nvme id-ctrl /dev/nvme0` and attach to those results?

keithbusch, 2018-01-17:
And assuming there is a /dev/nvme0 and if id-ctrl's 'nn' field says 1 or higher, also attach `nvme id-ns /dev/nvme0 -n 1`.

scorchpc, 2018-01-17:
I do not have a namespace (/dev/nvm*), and I think we found the issue, we need a newer kernel. Our Kernel worked for most NVME drives, but not some newer Plextor drives. Thanks for the quick response though! 

Closing this.

closed 2018-01-17 by scorchpc

scorchpc, 2018-01-17:
root@tsgal99:~ cat /proc/version
Linux version 3.8.1-tsl-v1.7.3 (anderson@anderson-probook) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-159-g6e7c616) ) 1 SMP PREEMPT Mon Jun 26 18:01:10 PDT 2017
root@tsgal99:~ uname -r
3.8.1-tsl-v1.7.3

And here is the config of the working device:
root@tsgal21:~ uname -r
4.9.76-tsl-v1.7.3+
root@tsgal21:~ cat /proc/version
Linux version 4.9.76-tsl-v1.7.3+ (anderson@anderson-probook) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-159-g6e7c616) ) 1 SMP PREEMPT Tue Jan 16 17:57:28 PST 2018

scorchpc, 2018-01-17:
Just to be clear, I had 3.8.1 kernel and the working is 4.9.76

We have our own version of linux (tsl-v.1.7.3) which I think is based on debian