linux-nvme/nvme-cli #284: big-endian systems

tim-oleksii, 2018-01-21:
Should the nvme-cli consider running on big-endian architectures? I stumbled upon code
https://github.com/linux-nvme/nvme-cli/blob/69b377995c0c05d297486f405fa54889355c8f22/wdc-nvme.c#L1212

However, I wonder if the plugin has a desire to preserve only 16 bytes (this is why it's converted to ushort after all), then why the mask is applied before conversion? Is it OK on BE architectures?

Thanks for clarification!

keithbusch, 2018-02-21:
It should support big-endian. It does look like the masking and endian conversion is backwards in this vendor specific extension, though.

tim-oleksii, 2018-02-21:
Yep, the code does it wrong way round. @jeffreyalien Could you please give us more insights? I can do a fix if it needs it.

jeffreyalien, 2018-02-21:
The reason for masking the nand_bad_block field is  because it's split into 2 parts.  2 bytes are for a normalized count and the 6 remaining bytes are for a raw count.  There are several other fields in this same data structure defined the same way and all have the same potential issue.  I will look into fixing the endian conversion and masking order.  Thanks for pointing this out.