linux-nvme/nvme-cli #288: Any roadmap for nvme-cli porting on ARM architectures? Is it currently possible compiling for arm?

melqu1ades, 2018-01-25:
I'm interested in using nvme-cli in development platforms featuring ARM processors (HiKey960 is an example).

keithbusch, 2018-01-25:
I don't have an ARM machine to test on, but I could swear it works there, or at least used to. What's the first problem you're encountering with it?

melqu1ades, 2018-01-29:
Hi Keith,
thank you for your feedback.
I've gone thorough the Makefile that comes with nvme-cli and I found no references to any ARM compilation.Can you give me directions about how to compile nvme-cli on ARM architecture?
If you are involved in nvme-cli development and you're interested to cover also ARM compilation, I can provide you with a HiKey960 board to validate your code.Consider that also Automotive market is switching to NVMe drives, and many infotainment systems are ARM based.
Thanks for your support,Guglielmo 

----Messaggio originale----

Data: 25-gen-2018 16.40

A: "linux-nvme/nvme-cli"<nvme-cli@noreply.github.com>

Cc: "melqu1ades"<guglielmo.ciavorella@tin.it>, "Author"<author@noreply.github.com>

tim-oleksii, 2018-01-29:
I was wondering whether you need to compile the tool in ARM host or just cross-compile in x86 PC into ARM executable?

keithbusch, 2018-01-29:
Are you building on the target machine, or cross compiling? If on target machine, just running 'make' should work. If cross compiling, you need to set up the environment before running 'make'.

melqu1ades, 2018-01-29:
Thank you.
The system I'm running experiment is the following:
Linux hikey960 4.14.0-rc7-linaro-hikey960 #1 SMP PREEMPT Fri Nov 24 10:45:34 UTC 2017 aarch64 aarch64 aarch64 GNU/Linux
I'm having troubles with nvme driver management: perhaps I need to recompile kernel.When solved I'll follow your suggestion.I'll be back to you in case of additional help need.
Thanks a lot to you all.
Regards,Guglielmo

----Messaggio originale----

Data: 29-gen-2018 16.58

melqu1ades, 2018-04-11:
Hi Keith,
here I am back to this task.
I apologize for the long lasting issue.
I have now two **Hikey960** that I use as follows:
- system (Hikey960) with Linaro Linux (my build system) where

closed 2018-05-23 by melqu1ades

wushgit, 2023-01-31:
Hi, @melqu1ades 

    I have a same requirement as you , need a nvme-cli tool can be used on an ARM board with Android. Can I have your solution?

igaw, 2023-01-31:
nvme-cli 2.x build on ARM. We have CI cross builds for this arch:

https://github.com/linux-nvme/nvme-cli/blob/137fb15a285c060789581fc074ab3b1f6b250188/.github/workflows/build.yml#L28

Obviously, the build depends on a proper distribution such Debian. Currently, we don't have any CI builds enabled for more embedded systems based on buildroot or Yocto.