linux-nvme/nvme-cli #304: [nvme-cli] nvme.c:4134 crash

tim-oleksii, 2018-02-12:
Stumbled upon an issue while flashing my SSD, haven't anaylzed it, but I think it's still needed as an issue

I'll get it analyzed tomorrow.

`root@test-Precision-WorkStation-T5500:~/upstream_nvme/nvme-cli# make nvme
NVME_VERSION = 1.5.100.g4d41
root@test-Precision-WorkStation-T5500:~/upstream_nvme/nvme-cli# ./nvme fw-download -f FW.bin /dev/nvme0n1
Firmware download success
*** Error in `./nvme': munmap_chunk(): invalid pointer: 0x00007efd2a577960 ***
Aborted (core dumped)
root@test-Precision-WorkStation-T5500:~/upstream_nvme/nvme-cli# gdb ./nvme
GNU gdb (Ubuntu 8.0.1-0ubuntu1) 8.0.1
Copyright (C) 2017 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./nvme...done.
(gdb) run fw-download -f FW.bin /dev/nvme0n1
Starting program: /root/upstream_nvme/nvme-cli/nvme fw-download -f FW.bin /dev/nvme0n1
Firmware download success
*** Error in `/root/upstream_nvme/nvme-cli/nvme': munmap_chunk(): invalid pointer: 0x00007ffff7fdd960 ***

Program received signal SIGABRT, Aborted.
__GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:51
51      ../sysdeps/unix/sysv/linux/raise.c: No such file or directory.
(gdb) bt
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:51
#1  0x00007ffff7828f5d in __GI_abort () at abort.c:90
#2  0x00007ffff787128d in __libc_message (action=action@entry=do_abort, fmt=fmt@entry=0x7ffff7998528 "*** Error in `%s': %s: 0x%s ***\n") at ../sysdeps/posix/libc_fatal.c:181
#3  0x00007ffff787f650 in malloc_printerr (ar_ptr=0x0, ptr=<optimized out>, str=0x7ffff7998840 "munmap_chunk(): invalid pointer", action=<optimized out>) at malloc.c:5426
#4  munmap_chunk (p=<optimized out>) at malloc.c:2877
#5  __GI___libc_free (mem=<optimized out>) at malloc.c:3138
#6  0x000055555555df29 in fw_download (argc=<optimized out>, argv=<optimized out>, cmd=<optimized out>, plugin=<optimized out>) at nvme.c:1949
#7  0x0000555555572369 in handle_plugin (argc=4, argv=0x7fffffffe4d0, plugin=0x555555792060 <builtin>) at plugin.c:158
#8  0x0000555555558296 in main (argc=5, argv=0x7fffffffe4c8) at nvme.c:4134
`

keithbusch, 2018-02-12:
ha, yah, i see the problem. We're freeing the incremented pointer, rather than original allocated pointer. will fix shortly.

tim-oleksii, 2018-02-12:
https://github.com/linux-nvme/nvme-cli/blob/4d417d931ceae3dcb2c9ba9306867f40de7e3004/nvme.c#L1941

Yep, it's quite easy to fix.

tim-oleksii, 2018-02-12:
Ok, just realized that you added that you're going to do it.

keithbusch, 2018-02-12:
Regression introduced with 7151acc on Jan 27

tim-oleksii, 2018-02-12:
> root@test-Precision-WorkStation-T5500:~/upstream_nvme/nvme-cli# make nvme
> NVME_VERSION = 1.5.101.g78b6
> cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -DLIBUUID -DNVME_VERSION='"1.5.101.g78b6"' nvme.c -o nvme argconfig.o suffix.o parser.o nvme-print.o nvme-ioctl.o nvme-lightnvm.o fabrics.o json.o plugin.o intel-nvme.o lnvm-nvme.o memblaze-nvme.o wdc-nvme.o wdc-utils.o nvme-models.o huawei-nvme.o -luuid
> root@test-Precision-WorkStation-T5500:~/upstream_nvme/nvme-cli# valgrind ./nvme fw-download -f FW.bin /dev/nvme0n1
> ==7025== Memcheck, a memory error detector
> ==7025== Copyright (C) 2002-2017, and GNU GPL'd, by Julian Seward et al.
> ==7025== Using Valgrind-3.13.0 and LibVEX; rerun with -h for copyright info
> ==7025== Command: ./nvme fw-download -f FW.bin /dev/nvme0n1
> ==7025==
> Firmware download success
> ==7025==
> ==7025== HEAP SUMMARY:
> ==7025==     in use at exit: 0 bytes in 0 blocks
> ==7025==   total heap usage: 32 allocs, 32 frees, 1,559,618 bytes allocated
> ==7025==
> ==7025== All heap blocks were freed -- no leaks are possible
> ==7025==
> ==7025== For counts of detected and suppressed errors, rerun with: -v
> ==7025== ERROR SUMMARY: 0 errors from 0 contexts (suppressed: 0 from 0)
>

closed 2018-02-12 by tim-oleksii