linux-nvme/nvme-cli #311: 'nvme id-ctrl' json output not valid

piyush-nimbalkar, 2018-02-23:
Was trying to use the output of `nvme id-ctrl /dev/nvmexx` in a script using the json format. The json output does not consider the newlines properly and returns an invalid output.

```
{
    "vid" : 4473,
    "ssvid" : 4473,
    "sn" : "0cc47ad37b71-3
",
    "mn" : "NVMF-XX-XXX
",
    "fr" : "0101
",
    "subnqn" : "nqn.1992-03.com.target
",
// rest of the fields
...
}
```

The above is not a valid json syntax.

I see the normal output does not get affected with newlines and there are no extra newlines after the above fields.

keithbusch, 2018-02-23:
Weird. If you're getting newlines in the json output, that should mean the controller returned a newline in that field. That would be a spec violation if the controller is doing that.

Below is output from my controller below, so it certainly doesn't look like the tool is appending those newlines.

```
# nvme id-ctrl /dev/nvme0 -o json
{
  "vid" : 32902,
  "ssvid" : 32902,
  "sn" : "BTLF732107UL1P0GGN  ",
  "mn" : "INTEL SSDPE2KX010T7                     ",
  "fr" : "QDV10130",
  "rab" : 0,
  "ieee" : 6083300,
...
}
```

piyush-nimbalkar, 2018-02-23:
Not saying that the tool is adding the newlines. It's just that `-o json` should return a valid json. 
Also, I am unsure why the normal output doesn't have the extra newlines.

```
NVME Identify Controller:
vid     : 0x1179
ssvid   : 0x1179
sn      : 0cc47ad37b71-3
mn      : NVMF-XX-XXX
fr      : 0101
rab     : 6
subnqn  : nqn.1992-03.com.subsystem1
ps    0 : mp:0.00W operational enlat:0 exlat:0 rrt:0 rrl:0
```

If we are trimming the fields in the above output, shouldn't it fine to trim in the json output as well?
Or at least have the '\n'?

Will look into the controller spec violation separately.

keithbusch, 2018-02-23:
What version of this tool are you using (nvme --version)?

piyush-nimbalkar, 2018-02-23:
`nvme version 1.2`

keithbusch, 2018-02-23:
Okay, please upgrade.

6d2f0c827f7b3d6283ba4b14ab0ba5382380993d

closed 2018-02-23 by keithbusch

piyush-nimbalkar, 2018-02-24:
Thanks!
Sorry for the inconvenience.