linux-nvme/nvme-cli #316: nvme-cli documentation build

Bm020813, 2018-03-08:
I'm having some issues with trying to build the documentation.  My goal is to eventually add new documentation to support a plug-in I'm working on for NVMe FC support.
When I go to build the docs with make doc, I'm getting the following errors when xmlto executes.
 (output is abbreviated, as the list is very long):
xmlto -m manpage-normal.xsl   man nvme-attach-ns.xml
I/O error : Attempt to load network entity http://docbook.sourceforge.net/release/xsl/current/common/entities.ent
http://docbook.sourceforge.net/release/xsl/current/html/autoidx.xsl:4: warning: failed to load external entity "http://docbook.sourceforge.net/release/xsl/current/common/entities.ent"
%common.entities;
                 ^
Entity: line 1:
 %common.entities;
                  ^
http://docbook.sourceforge.net/release/xsl/current/html/autoidx.xsl:33: parser error : Entity 'primary' not defined
         use="translate(substring(&primary;, 1, 1),&lowercase;,&uppercase;)"/>
                                           ^
http://docbook.sourceforge.net/release/xsl/current/html/autoidx.xsl:33: parser error : Entity 'lowercase' not defined
         use="translate(substring(&primary;, 1, 1),&lowercase;,&uppercase;)"/>
                                                              ^
http://docbook.sourceforge.net/release/xsl/current/html/autoidx.xsl:33: parser error : Entity 'uppercase' not defined
         use="translate(substring(&primary;, 1, 1),&lowercase;,&uppercase;)"/>
                                                                          ^
http://docbook.sourceforge.net/release/xsl/current/html/autoidx.xsl:37: parser error : Entity 'primary' not defined
         use="&primary;"/>
                       ^
http://docbook.sourceforge.net/release/xsl/current/html/autoidx.xsl:41: parser error : Entity 'primary' not defined
         use="concat(&primary;, &sep;, &secondary;)"/>
                              ^
http://docbook.sourceforge.net/release/xsl/current/html/autoidx.xsl:41: parser error : Entity 'sep' not defined
         use="concat(&primary;, &sep;, &secondary;)"/>
                                     ^
It seems that the http://docbook.sourceforge.net files are missing entries or this makefile is out-of-date with the latest files on this site.   Any insight would be helpful to resolve this.  Thanks.

keithbusch, 2018-03-08:
Hmm ... I have to admit the documentation Makefile was copied from 'git' (there's tons of unnecessary artifacts in there that have nothing to do with this nvme project), and I never bothered to look into the inner workings.

Seems to be working fine on all my machines, so sounds like your xmlto pacakge might be pointing to something invalid,, but I'd also rather this be able to work offline without such dependencies. I'll have to get back to you on this... If you want to just commit the .txt file, I periodically regenerate all the documentation.

Bm020813, 2018-03-08:
Hi Keith,

I manually generated the manpage and html files for now using txt2man and
asciidoc tools.  When I fork should I check those in as well?  Or just the
.txt files?

I got it building and creating an RPM  package.  I haven't tried the debian
builds as I don't have that OS installed.

However, there seems to be an issue with installing the generated RPM on
SuSE SLES 12 SP3.
The RPM spec file has a requirement for uuidgen and this fails on SLES 12
SP3.  I cannot find the package that resolves this requirement,  so I
locally removed that line in the spec file for now to test it.
The uuidgen tool is installed but RPM can't find that package so it may be
part of another RPM package on SLES.  Also there is no libuuid-devel
package for SLES 12.

Finally, is there a requirement to support .json output or is that optional
for a plug-in that prints to the console?

FY_ - Here is a sample of our command output so far:

 # nvme brcm-fc list-ns
Port    Namespace  Subsystem NQN               Capacity  BlkSz
---------------------------------------------------------------------------
lpfc0   nvme0n1    nqn.2016-06.io.spdk:cnode1  67 MB    512
lpfc0   nvme0n2    nqn.2016-06.io.spdk:cnode1  67 MB    512
lpfc0   nvme1n1    nqn.2016-06.io.spdk:cnode1  67 MB    512
lpfc0   nvme1n2    nqn.2016-06.io.spdk:cnode1  67 MB    512
lpfc1   nvme0n1    nqn.2016-06.io.spdk:cnode1  67 MB    512
lpfc1   nvme0n2    nqn.2016-06.io.spdk:cnode1  67 MB    512
lpfc1   nvme1n1    nqn.2016-06.io.spdk:cnode1  67 MB    512
lpfc1   nvme1n2    nqn.2016-06.io.spdk:cnode1  67 MB    512
lpfc2   nvme0n1    nqn.2016-06.io.spdk:cnode1  67 MB    512
lpfc2   nvme0n2    nqn.2016-06.io.spdk:cnode1  67 MB    512
lpfc2   nvme1n1    nqn.2016-06.io.spdk:cnode1  67 MB    512
lpfc2   nvme1n2    nqn.2016-06.io.spdk:cnode1  67 MB    512
lpfc3   nvme0n1    nqn.2016-06.io.spdk:cnode1  67 MB    512
lpfc3   nvme0n2    nqn.2016-06.io.spdk:cnode1  67 MB    512
lpfc3   nvme1n1    nqn.2016-06.io.spdk:cnode1  67 MB    512
lpfc3   nvme1n2    nqn.2016-06.io.spdk:cnode1  67 MB    512

linux-wih8:~ # nvme brcm-fc list-conn
Port    Connection
                                        Subsystem NQN               D_ID
  Status
--------------------------------------------------------------------------------------------------------------------------------------------------------------------
lpfc0
 traddr=nn-0x20000090faf08ed4:pn-0x10000090faf08ed4,host_traddr=nn-0x20000090fa9483c9:pn-0x10000090fa9483c9
nqn.2016-06.io.spdk:cnode1  0x010e00  Online
lpfc0
 traddr=nn-0x20000090faf08ed4:pn-0x10000090faf08ed4,host_traddr=nn-0x20000090fa9483c8:pn-0x10000090fa9483c8
nqn.2016-06.io.spdk:cnode1  0x010e00  Online
lpfc1
 traddr=nn-0x20000090faf08ed4:pn-0x10000090faf08ed4,host_traddr=nn-0x20000090fa9483c9:pn-0x10000090fa9483c9
nqn.2016-06.io.spdk:cnode1  0x010600  Online
lpfc1
 traddr=nn-0x20000090faf08ed4:pn-0x10000090faf08ed4,host_traddr=nn-0x20000090fa9483c8:pn-0x10000090fa9483c8
nqn.2016-06.io.spdk:cnode1  0x010600  Online
lpfc2
 traddr=nn-0x20000090faf08ed4:pn-0x10000090faf08ed4,host_traddr=nn-0x20000090fa9483c9:pn-0x10000090fa9483c9
nqn.2016-06.io.spdk:cnode1  0x010700  Online
lpfc2
 traddr=nn-0x20000090faf08ed4:pn-0x10000090faf08ed4,host_traddr=nn-0x20000090fa9483c8:pn-0x10000090fa9483c8
nqn.2016-06.io.spdk:cnode1  0x010700  Online
lpfc3
 traddr=nn-0x20000090faf08ed4:pn-0x10000090faf08ed4,host_traddr=nn-0x20000090fa9483c9:pn-0x10000090fa9483c9
nqn.2016-06.io.spdk:cnode1  0x010f00  Online
lpfc3
 traddr=nn-0x20000090faf08ed4:pn-0x10000090faf08ed4,host_traddr=nn-0x20000090fa9483c8:pn-0x10000090fa9483c8
nqn.2016-06.io.spdk:cnode1  0x010f00  Online

linux-wih8:~ # nvme brcm-fc list-adp
Port    Description                                                 Host
WWPN           Host WWNN           S/N                     Driver Ver.
   FW Rev.
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------
lpfc0   Emulex LPe35004-S3-IS 4-Port 32Gb Fibre Channel Adapter
 0x10000090fa9483c6  0x20000090fa9483c6  NP80200250
12.0.114.0          12.0.114.0
lpfc1   Emulex LPe35004-S3-IS 4-Port 32Gb Fibre Channel Adapter
 0x10000090fa9483c7  0x20000090fa9483c7  NP80200250
12.0.114.0          12.0.114.0
lpfc2   Emulex LPe35004-S3-IS 4-Port 32Gb Fibre Channel Adapter
 0x10000090fa9483c8  0x20000090fa9483c8  NP80200250
12.0.114.0          12.0.114.0
lpfc3   Emulex LPe35004-S3-IS 4-Port 32Gb Fibre Channel Adapter
 0x10000090fa9483c9  0x20000090fa9483c9  NP80200250
12.0.114.0          12.0.114.0

Thanks for the help.
-Bruce
*____________________________*
*Bruce Maynard*
*Broadcom *ECD | R & D
office: 949-926-1114 <(949)%20926-1114>
15101 Alton Pkwy., Irvine, CA 92618
bruce.maynard@broadcom.com

keithbusch, 2018-03-08:
The distros don't use the spec files here for their packaging, so don't worry much about that.

When it comes to vendor extensions, I don't enforce rules on program output. If you don't have a use for json, feel free to omit a mode for it. It can always be added later if the desire arises.

I'll just mention, though, the primary reason for the vendor extensions is to support non-standard commands. These functions at least appear pretty useful to any nvme over fibre channel. If it's just using NVMe identifies, it'd be nice to make these generic commands; if it's truly using Broadcom-specifics, that's cool too.

igaw, 2022-01-19:
No progess in this bug report for a long time. If you think it's still relevant please reopen.

closed 2022-01-19 by igaw