linux-nvme/nvme-cli #317: Use case for nvme subsystem-reset

narasimhan-v, 2018-03-14:
Hi Keith,

Can you please explain what a typical user would use `nvme subsystem-reset` for.

keithbusch, 2018-03-14:
The spec provides a few use cases. One being in response to a return code for f/w activation, response 110h: "Firmware Activation Requires NVM Subsystem Reset".

I am aware Linux may have a hard time dealing with this. This type of reset directs the links to LTSSM. This can cause drives to appear unresponsive if the slot doesn't report link states, or even if it does, could get confused if the link state changes rapidly enough.

narasimhan-v, 2018-03-15:
Thanks Keith, for the explanation. I will look into spec for more details.
And yes, we are seeing many drives disappearing as well.

tim-oleksii, 2018-03-15:
btw fw-activate points out in documentation that it needs some kind of reset.
as an example "echo 1 > /...nvme/reset" is listed.

Whenever I tried this approach it didn't worked. (ubuntu, centos).

keithbusch, 2018-03-15:
? It better work. What happened instead?

tim-oleksii, 2018-03-15:
OK, so it should, as a work around I used 'ctrl-reset'. I wondered whether it had been FW issue or kernel.

Let me see FW logs first, in case it's OK is dmesg enough?

keithbusch, 2018-03-15:
Let's establish what's going on first. Please answer the following:

What did you do?
What did you expect to happen?
What actually happened?

tim-oleksii, 2018-03-15:
1. I tried to update FW (fw-download, fw-commit, echo 1 > reset).

Here are excerpts from shell
> [root@rhel72 test]# ./nvme-cli/nvme fw-download /dev/nvme0 -f ./FW.bin
> Firmware download success
> [root@rhel72 test]# ./nvme-cli/nvme fw-commit /dev/nvme0 -s 1 -a 1
> Success committing firmware action:1 slot:1
> [root@rhel72 test]# echo 1 > /sys/class/nvme/nvme0/device/reset
> [root@rhel72 test]# uname -a
> Linux rhel72.sfstool 3.10.0-327.el7.x86_64 #1 SMP Thu Oct 29 17:29:29 EDT 2015 x86_64 x86_64 x86_64 GNU/Linux
> [root@rhel72 test]# ls /dev | grep nvme
> nvme0
> nvme0n1
> [root@rhel72 test]# ./nvme-cli/nvme id-ctrl /dev/nvme0
> /dev/nvme0: No such device
> [root@rhel72 test] dmesg
> [  155.228816] nvme 0000:04:00.0: async event result 00030102
> [  194.902604] nvme 0000:04:00.0: Device not ready; aborting initialisation

2. Expected behaviour is the OK sequence that can be reproduced as fw-download, fw-commit, reset (all are nvme-cli commands).
3. It's seen that device went missing. No FW crash reported, id-ctrl failed.

PS: at least documentation is incosistent (however, I'm not the one from LKM to judge whether it's correct or not).

https://github.com/linux-nvme/nvme-cli/blob/242012d1111817542582ff861d9507aa90f95e10/Documentation/nvme-fw-commit.txt#L35

https://github.com/linux-nvme/nvme-cli/blob/186a72433b45ce34022c61e167995ba34bcc544d/nvme.c#L2024

keithbusch, 2018-03-15:
The first will usually send a PCIe FLR, the second will just do a nvme controller reset. While the FLR does way more than necessary (typically), either should be valid according to spec. If your device doesn't support FLR, an even more harsh reset method could happen, escalating up to a secondary bus reset.

As far as kernel support goes, wiring up the FLR notification to drivers wasn't added till well after 3.10, and the 7 elrepo wouldn't pick it up since it breaks kABI. Looks like the devs back-ported this feature some other way, but I've never tested it.

In any case, it looks like your device encounters an internal error using the PCI reset method. I would avoid using that method anyway since it's a bit harsh. You can use either the nvme specific "reset_controller" sysfs or the NVME_IOCTL_RESET.

tim-oleksii, 2018-03-15:
Yes, I use nvme-cli reset mainly to avoid this malfunction.
Thank you for clarification. I don't mind of closing the issue.

closed 2018-03-15 by keithbusch

akshimmu, 2023-12-06:
If the default lbaf for a device is 4k+0 after boot up and then it is formatted to 512+0 and then a nssr is triggered, is it expected to go back to 4k+0 or remain at 512+0?