linux-nvme/nvme-cli #318: Sending a fused cmd via nvme cli

RaghuAdabala, 2018-03-15:
I need some help with issuing fused cmds (specifically and compare() and write()).
The nvme io-passthru interface does not seem to allow me to set the 'fuse' bit.
Is there some other way to issue  fused cmds ?.
TIA,
Raghu

keithbusch, 2018-03-15:
The kernel doesn't support it, so we can't do it from user space.

RaghuAdabala, 2018-03-15:
Thanks Keith !.
Any ideas as to when Linux will support fused cmds ?
Regards,
Raghu.

keithbusch, 2018-03-15:
No current plans, as far as I know. No one has ever asked for it, and there's no in-kernel use for it either, so it's never been on the development radar.

This would almost certainly require an entire new ioctl if there really is a need to issue fused commands from user space.

keithbusch, 2018-07-12:
That's not going to work. The ioctl is a blocking call, and you can't get a response to the first fuse until you post the second, so now you need to send each part in two carefully coordinated threads that you must ensure are scheduled on the same CPU... but then there's also nothing you can do from those user threads to ensure that an unrelated command doesn't slip in-between your two fused operations.

rolandd, 2018-07-12:
Oh, good point, my mistake.

I guess we need a new "submit fused command" ioctl, which wouldn't be too hard to implement.

I'll take a stab at that, as well as nvme-cli tooling, since I am planning on implementing fused command handling in my target and I need a good way to test.

keithbusch, 2018-07-12:
I definitely support the cause, so thanks for volunteering. :)

I'm not seeing a convenient way to make sure the necessary sequence will happen given the existing blk-mq interfaces, so I'm not sure the new ioctl will be completely straight forward unless you see a clever solution I missed. I would be happy to help review any driver/kernel proposals, though.

rolandd, 2018-07-12:
We can put whatever we want in the req->cmd pointer, including a pointer to a buffer with two command descriptors.  So it doesn't really support kernel consumers the way a REQ_OP_COMPARE_AND_WRITE might, but I don't see a fundamental issue plumbing it through.

keithbusch, 2018-07-12:
I hadn't thought about using a single-request for a fused command because of the problems when you're trying to create the request's payload and then map it to the two command PRPs/SGLs: the driver wouldn't be able know where to split blk_rq_payload_bytes without decoding command specific fields, so you can't generically enable fused operations this way.

This also becomes even more problematic if the two fuse operations have data transfers going in opposite directions, since we don't have a way to describe multi-directional payloads within a single request (compare+write doesn't have that problem).

We should probably take this to linux-nvme mailing list.

kennethburgener, 2019-07-10:
Curious, is there a link to where this thread continued on in the linux-nvme mailing list?

ranjanchoubey, 2020-07-21:
Hi @keithbusch , @rolandd : Any latest progress on the support of fused commands that you are aware of ?

keithbusch, 2020-07-21:
No one has brought any proposals to the kernel mailing list, so no update.

closed 2022-01-19 by igaw