linux-nvme/nvme-cli #324: Cant get my NVMe device back on `nvme list` after deleting namespace

kreuzerkrieg, 2018-03-22:
I've deleted a namespace on my NVMe device, then, I created it again but I dont see the device on `nvme list` nor as block device /dev/nvmeXn1. 

delete/create

> # nvme delete-ns /dev/nvme10 -n 1
> delete-ns: Success, deleted nsid:1
> # nvme create-ns /dev/nvme10 -s 1562824368 -c 1562824368 -flbas 0
> create-ns: Success, created nsid:1

format

># nvme format /dev/nvme10 -n 1 -l 0 -r
>NVME Admin command error:INVALID_FIELD(6002)

or

># nvme format /dev/nvme10 -l 0 -r
>Success formatting namespace:ffffffff

Not sure I have to format the device after creating the namespace...
Should I attach the ns to controller first?

># nvme attach-ns /dev/nvme10 -n 1 -c 35,36
>NVMe Status:NS_IS_PRIVATE(6119)

Whats wrong with what I'm doing?

kreuzerkrieg, 2018-03-22:
Looks like I've found the problem, cant attach to a list of controllers. If I attach to controllers one by one, it works

> nvme delete-ns /dev/nvme10 -n 1
> nvme create-ns /dev/nvme10 -s 1562824368 -c 1562824368 -f 0 -d 0 -m 1
> nvme attach-ns /dev/nvme10 -n 1 -c 35
> nvme attach-ns /dev/nvme10 -n 1 -c 36
> nvme format /dev/nvme10 -n 1

Bug?

keithbusch, 2018-03-22:
In your first example, you didn't have "-m" option, which is required if you intend to share the namespace among multiple controllers.

kreuzerkrieg, 2018-03-22:
So, this is the only problem? `-m 1`? If so, my apologies for false alarm

keithbusch, 2018-03-22:
Yes, that appears to be the only problem. If you intend to attach it to just a single controller, you don't need to set that option.

kreuzerkrieg, 2018-03-22:
Thanks for clarification

closed 2018-03-22 by kreuzerkrieg