linux-nvme/nvme-cli #325: The use of CDW11 parameter in the get-feature command

rrosatti, 2018-03-23:
The CDW11 is used for interrupt vector config. So, is there a way for me to know the "possible values" I can use on this parameter for a specific adapter?

For example, in one adapter I could use the values from 0 to 18:

[root@ltc-zz4-lp13 nvme-cli]# ./nvme get-feature /dev/nvme0 -f 0x09 --cdw11=0
get-feature:0x9 (Interrupt Vector Configuration), Current value:0x010000
[root@ltc-zz4-lp13 nvme-cli]# ./nvme get-feature /dev/nvme0 -f 0x09 --cdw11=18
get-feature:0x9 (Interrupt Vector Configuration), Current value:0x000012
[root@ltc-zz4-lp13 nvme-cli]# ./nvme get-feature /dev/nvme0 -f 0x09 --cdw11=19
NVMe Status:INVALID_FIELD: A reserved coded value or an unsupported value in a defined field(2002)

But, with other one I could use a range from 0 to 128:

[root@ltc-briggs6 nvme-cli]# ./nvme get-feature /dev/nvme0 -f 0x09 --cdw11=0
get-feature:0x9 (Interrupt Vector Configuration), Current value:0x010000
[root@ltc-briggs6 nvme-cli]# ./nvme get-feature /dev/nvme0 -f 0x09 --cdw11=128
get-feature:0x9 (Interrupt Vector Configuration), Current value:0x010080
[root@ltc-briggs6 nvme-cli]# ./nvme get-feature /dev/nvme0 -f 0x09 --cdw11=129
NVMe Status:INVALID_FIELD: A reserved coded value or an unsupported value in a defined field(2)

Thanks in advance,
Rodrigo

keithbusch, 2018-03-23:
The max value here is really tied to the highest enabled MSI-x vector on the device, which you can check with 'lspci' on your controller.

Another way with Linux is to check the queue count since the driver does these 1:1. First run:

```
# nvme get-feature /dev/nvme0 -f 7
get-feature:0x7 (Number of Queues), Current value:0x7f007f
```
The result is two words: mask off the smaller value of the two (both the same in the above example) then add 1 (zeroes based values): 80h == 128, and that's the highest value you can set for interrupt vector configuration for the above.

keithbusch, 2018-03-23:
Another way: `# grep nvme0 /proc/interrupts | wc -l`

rrosatti, 2018-03-23:
Thanks for this information, Keith!

Just for curiosity, in one of the adapters all the 3 things you proposed to do showed the same result.
But to the other one, just the 'lspci' one seems to be correct.

[root@ltc-zz4-lp13 ~]# grep nvme0 /proc/interrupts | wc -l
8
[root@ltc-zz4-lp13 ~]# nvme get-feature /dev/nvme0 -f 7
get-feature:0x7 (Number of Queues), Current value:0x0f000f
[root@ltc-zz4-lp13 ~]# lspci -vv | grep -i msi-x
	Capabilities: [b0] MSI-X: Enable+ Count=19 Masked-

For the above, the maximum value I could use was '18'. But, the '/proc/interrups' showed 8 and get-feature (Number of Queues) showed 16.

Well, 'lspci' seems to be the "safest" one to use =)

keithbusch, 2018-03-23:
The output from /proc/interrupts is the one that actually matters since those are the only vectors the host can actually use. The others are beyond the range of what the host has allocated so while you can toggle their settings, it doesn't change anything meaningful.

rrosatti, 2018-03-23:
Got it. Thanks a lot for the information!

closed 2018-03-26 by keithbusch