linux-nvme/nvme-cli #329: nvme crash with nvme io-passthru command

UmaBankolli, 2018-03-29:
double free in nvme command with nvme io-passthru comamnd, 
nvme io-passthru /dev/nvme0n1 -n 1 -o 0x2 --cdw10=0 --cdw12=0x1 -l 512  -r -i fileR
bt:
(gdb) bt
#0  0x00007eff2cc201d7 in raise () from /lib64/libc.so.6
#1  0x00007eff2cc218c8 in abort () from /lib64/libc.so.6
#2  0x00007eff2cc5ff07 in __libc_message () from /lib64/libc.so.6
#3  0x00007eff2cc67503 in _int_free () from /lib64/libc.so.6
#4  0x0000000000405fd7 in passthru (argc=<optimized out>, argv=<optimized out>, ioctl_cmd=<optimized out>, desc=<optimized out>, cmd=<optimized out>) at nvme.c:3699
#5  0x0000000000416af0 in handle_plugin (argc=argc@entry=13, argv=argv@entry=0x7fff2df4fb60, plugin=0x632280 <builtin>) at plugin.c:150
#6  0x00000000004017e1 in main (argc=14, argv=0x7fff2df4fb58) at nvme.c:3781
(gdb)
==========================other info==================
1.
[ nvme-cli-1.5]# nvme io-passthru /dev/nvme0n1 -n 1 -o 0x2 --cdw10=0 --cdw12=0x0 -l 512  -r -i fileR
NVMe command result:00000000
       0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
0000: bd d9 4a bf 6f 6f 49 1a 9a 3b 49 1a 97 57 cd 8d "..J.ooI..;I..W.."
0010: 44 1f d8 c1 5f bf 0f 43 40 cc 44 bb 3e c0 c3 00 "D..._..C@.D.>..."
0020: 7c 79 83 7d 17 02 ca f2 0b 31 54 aa cf 60 2d d0 "|y.}.....1T..`-."
0030: 3f 8e 48 53 db a7 6a 41 82 ba f0 f0 b6 e1 c0 9e "?.HS..jA........"
:
:
[ nvme-cli-1.5]#
2. 
[ nvme-cli-1.5]# nvme io-passthru /dev/nvme0n1 -n 1 -o 0x2 --cdw10=0 --cdw12=0x1 -l 512  -r -i fileR
NVMe command result:00000000
       0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
0000: bd d9 4a bf 6f 6f 49 1a 9a 3b 49 1a 97 57 cd 8d "..J.ooI..;I..W.."
0010: 44 1f d8 c1 5f bf 0f 43 40 cc 44 bb 3e c0 c3 00 "D..._..C@.D.>..."
0020: 7c 79 83 7d 17 02 ca f2 0b 31 54 aa cf 60 2d d0 "|y.}.....1T..`-."
0030: 3f 8e 48 53 db a7 6a 41 82 ba f0 f0 b6 e1 c0 9e "?.HS..jA........"
0040: 87 4a 06 e3 2b f5 e3 f8 56 f6 f0 9d 51 a9 14 d9 ".J..+...V...Q..."
0050: 4a 2c f1 5c 6d 60 02 c4 61 22 44 32 97 79 3e 53 "J,.\m`..a"D2.y>S"
:
:
01e0: 0d 8e 11 ed cc 6d 5a c5 ac 2a 88 91 a0 9f ae 9f ".....mZ..*......"
01f0: eb 34 73 3e 65 93 e4 1f 7c 94 ba 25 19 bb 0f ff ".4s>e...|..%...."
*** Error in `nvme': double free or corruption (!prev): 0x0000000000ba7000 ***

======= Backtrace: =========
/lib64/libc.so.6(+0x7c503)[0x7eff2cc67503]
nvme[0x405fd7]
nvme[0x416af0]
nvme[0x4017e1]
/lib64/libc.so.6(__libc_start_main+0xf5)[0x7eff2cc0cb35]
nvme[0x401883]
======= Memory map: ========
00400000-00432000 r-xp 00000000 08:01 5526058                            /usr/local/sbin/nvme
00631000-00632000 r--p 00031000 08:01 5526058                            /usr/local/sbin/nvme
00632000-00634000 rw-p 00032000 08:01 5526058                            /usr/local/sbin/nvme
00ba6000-00bc7000 rw-p 00000000 00:00 0                                  [heap]
7eff20000000-7eff20021000 rw-p 00000000 00:00 0
7eff20021000-7eff24000000 ---p 00000000 00:00 0
7eff264ac000-7eff264c1000 r-xp 00000000 08:01 5505043                    /usr/lib64/libgcc_s-4.8.5-20150702.so.1
7eff264c1000-7eff266c0000 ---p 00015000 08:01 5505043                    /usr/lib64/libgcc_s-4.8.5-20150702.so.1
7eff266c0000-7eff266c1000 r--p 00014000 08:01 5505043                    /usr/lib64/libgcc_s-4.8.5-20150702.so.1
7eff266c1000-7eff266c2000 rw-p 00015000 08:01 5505043                    /usr/lib64/libgcc_s-4.8.5-20150702.so.1
7eff266c2000-7eff2cbeb000 r--p 00000000 08:01 5899735                    /usr/lib/locale/locale-archive
7eff2cbeb000-7eff2cda1000 r-xp 00000000 08:01 5505939                    /usr/lib64/libc-2.17.so
7eff2cda1000-7eff2cfa1000 ---p 001b6000 08:01 5505939                    /usr/lib64/libc-2.17.so
7eff2cfa1000-7eff2cfa5000 r--p 001b6000 08:01 5505939                    /usr/lib64/libc-2.17.so
7eff2cfa5000-7eff2cfa7000 rw-p 001ba000 08:01 5505939                    /usr/lib64/libc-2.17.so
7eff2cfa7000-7eff2cfac000 rw-p 00000000 00:00 0
7eff2cfac000-7eff2cfcc000 r-xp 00000000 08:01 5505932                    /usr/lib64/ld-2.17.so
7eff2d1b3000-7eff2d1b6000 rw-p 00000000 00:00 0
7eff2d1c8000-7eff2d1cb000 rw-p 00000000 00:00 0
7eff2d1cb000-7eff2d1cc000 r--p 0001f000 08:01 5505932                    /usr/lib64/ld-2.17.so
7eff2d1cc000-7eff2d1cd000 rw-p 00020000 08:01 5505932                    /usr/lib64/ld-2.17.so
7eff2d1cd000-7eff2d1ce000 rw-p 00000000 00:00 0
7fff2df31000-7fff2df52000 rw-p 00000000 00:00 0                          [stack]
7fff2dfc6000-7fff2dfc8000 r-xp 00000000 00:00 0                          [vdso]
ffffffffff600000-ffffffffff601000 r-xp 00000000 00:00 0                  [vsyscall]
Aborted (core dumped)
[root@jaguar32 nvme-cli-1.5]# nvme io-passthru /dev/nvme0n1 -n 1 -o 0x2 --cdw10=0 --cdw12=0x1 -l 1024  -r -i fileR
NVMe command result:00000000
       0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
0000: bd d9 4a bf 6f 6f 49 1a 9a 3b 49 1a 97 57 cd 8d "..J.ooI..;I..W.."
0010: 44 1f d8 c1 5f bf 0f 43 40 cc 44 bb 3e c0 c3 00 "D..._..C@.D.>..."
0020: 7c 79 83 7d 17 02 ca f2 0e 68 a3 99 10 74 d9 d1 ".......=.h...t.."
:
:
:
0140: d8 16 0b 89 d1 3d fa 70 20 70 72 6f 67 72 61 6d "his.nvme.program"
03d0: 20 69 73 20 61 20 75 73 65 72 20 73 70 61 63 65 ".is.a.user.space"
03e0: 20 75 74 69 6c 69 74 79 20 74 6f 20 70 72 6f 76 ".utility.to.prov"
03f0: 69 64 65 20 73 74 61 6e 64 61 72 64 73 20 63 6f "ide.standards.co"
[ nvme-cli-1.5]#

minwooim, 2018-03-29:
What is the LBA size of nvme0n1?
It can be found the following command.
```bash
# nvme id-ns /dev/nvme0n1 -H
...
LBA Format  0 : Metadata Size: 0   bytes - Data Size: 512 bytes - Relative Performance: 0 Best (in use)
```

It seems you may have the namespace 1 with LBA size as 512bytes.
If so, those given commands may trigger memory corruption because of a size mismatch between the --data-len=512 and --cdw12=0x1.
nvme-cli now prepares the data buffer for the read data just for given amount of --data-len value.

Those commands are all Read NVM command, and cdw12 should be zero-based value.
So, If you want to receive the data from the device only 512bytes, then you _should_ set it to 0x0 instead of 0x1.
That's why the first and third that you had given commands were successfully done.

```bash
# [PASS] nvme io-passthru /dev/nvme0n1 -n 1 -o 0x2 --cdw10=0 --cdw12=0x0 -l 512  -r -i fileR
# [FAIL] nvme io-passthru /dev/nvme0n1 -n 1 -o 0x2 --cdw10=0 --cdw12=0x1 -l 512  -r -i fileR
# [PASS] nvme io-passthru /dev/nvme0n1 -n 1 -o 0x2 --cdw10=0 --cdw12=0x1 -l 1024  -r -i fileR
```

If you're not familiar with your namespace, then it would be  great to use following built command.
```bash
# nvme read /dev/nvme0n1 -s 0 -c 0x0 -z 512 -d fileR
```

UmaBankolli, 2018-03-29:
Hi Minwoo,

Its the NS is of 4K block size. 
I was trying the various NLB, SLBA and datasize values and i hit this corruption. 

The main cause for the double free issue is that the allocated data buffer length is shorter than the length described in nlb, when you use 'nvme write' it allocates a buffer in the right size, unlike 'nvma io-passthru'  that allocates buffer according to -l.

thats why i had to raise a bug , though its a -ve case.

Thanks,
Uma

minwooim, 2018-03-29:
Oh, I see what your point is.
As you mentioned, the **io-passthru** command will prepare the data buffer in given value of -l.

But I'm not sure why the first one and the third that you had given seemed successfully done.
The buffer size would be 512bytes and the real size would be 4096bytes from the device.
I think it seems not in proper situation. What do you think about this?

Thanks,

UmaBankolli, 2018-03-29:
Hey Minwoo

read my above comments with BS=512, then it will be clear :)

Thanks,
Uma

minwooim, 2018-03-29:
I thought you have told me that your NS is of 4K block size instead of 512.
So you mean the test result that you posted is case of 512bytes block size?

Am I going well? If not, sorry in advance :(

keithbusch, 2018-03-29:
The NLB value is a "zeroes based value". If you set it to 1, you are requesting the device transfer 2 blocks, and you are overrunning the buffer you allocated for it.

closed 2018-03-30 by keithbusch

UmaBankolli, 2018-04-02:
This is a -ve case which crashes the nvme process. Its better to validate the inputs than simply let the process crash

keithbusch, 2018-04-02:
I disagree. The purpose of the passthrough is to do exactly what you ask it to do. It's not going to stop you from shooting yourself in the foot.