linux-nvme/nvme-cli #331: Incorrect value of Intel "time_workload_media _wear" counter when printed in json format

Shadiasd, 2018-04-02:
Hi,

I noticed that the raw value of the Intel "timed_workload_media_wear" counter is incorrect when printed in json format. The counter shows the correct value if printed as the default unformatted output. 

**Information required to reproduce the bug:**
- nvme_cli command: `sudo nvme intel  smart-log-add /dev/nvme0 -j`
- nvme_cli version: 1.3
- Please note that in case your drive media_wear is still zero, you might need to hardcode a random float value with 3 digit precision, such as "1.234", in your code as a raw value for "timed_workload_media_wear" to be able to reproduce this issue, 

**A potential fix:**
I was able to fix this issue, by changing the following lines in "nvme-intel.c" and json.c" files. Code below shows the modified version:

- nvme-intel.c, line 130 (changing the type cast from `float` to `long double`. This is necessary because `json_print_value` function for `JSON_TYPE_FLOAT` values uses `%Lf` as the printf format specifier):
`json_object_add_value_float(entry_stats, "raw", ((long double)int48_to_long(smart->timed_workload_media_wear.raw)) / 1024 );`

- json.c, line 377 (allowing up to 3 digit precision as opposed to none. This follows the same format that is used in function `show_intel_smart_log` (line 201) to print this counter)
`printf( "%.3Lf", value->float_number);`

However, I do not believe that this is a proper fix, as it changes the format of all the other numbers that are printed using `json_print_value` as a `JSON_TYPE_FLOAT`. 

It would be great if you can fix this issue.

Thank you,
Shadi

keithbusch, 2018-04-05:
should be fixed now

closed 2018-04-05 by keithbusch

Shadiasd, 2018-04-16:
Thank you for the fix! :)