linux-nvme/nvme-cli #332: Firmware slot detection

narasimhan-v, 2018-04-03:
Can someone please let me know if there is a way to detect firmware slots ?
I can see that there are 3 slots supported, but only one is seen.

```
# nvme fw-log /dev/nvme1
Firmware Log for device:nvme1
afi  : 0x11
frs1 : 0x51334234414e5047 (GPNA4B3Q)
#

# nvme fw-download /dev/nvme1 --fw=<image>
Firmware download success
# nvme fw-activate /dev/nvme1 -a 0 -s 1
NVME Admin command error:FIRMWARE_SLOT(106)
 
# nvme id-ctrl /dev/nvme1 -H | grep Firmware
[4:4] : 0x1   Firmware Activate Without Reset Supported
[3:1] : 0x3   Number of Firmware Slots
[0:0] : 0x1   Firmware Slot 1 Read-Only
```

keithbusch, 2018-04-03:
Your controller says firmware slot 1 is read only, and you're trying to activate a new image into that read-only slot.

keithbusch, 2018-04-03:
If you're asking why fw-log doesn't show the other slots, it's because you don't have anything valid in those slots for the program to show you.

narasimhan-v, 2018-04-04:
@keithbusch That is right, It is read only. But I have seen in controllers where they only show one slot, and when we try to fw-activate on the other slots, they become visible. So, is that the way ?

keithbusch, 2018-04-04:
The program is just the messenger. If you want to see firmware in any particular available slot, you need to put firmware in that slot so the device has something to report that the program can show you.

closed 2018-04-05 by keithbusch

got777888, 2021-07-26:
[image]
Tolong Bantu saya untuk Memecahkan masalah login ini, Saya tidak mengetahui user name dan Password nya