linux-nvme/nvme-cli #333: nvme get-log offset issue

blguerke, 2018-04-05:
Setting an offset doesn't seem work, e.g. pulling power log and reading a specific register setting:
nvme get-log /dev/nvme0n1 -i 0xc5 -l 0x4 -o 0x4
thnx

minwooim, 2018-04-17:
Hi blguerke,

As per my machine (linux-4.16 with nvme-cli latest version head ea195e85f35673c346c60c0386ddb2f25338b665) has a trace below.
#### Commnad processed
```bash
# nvme get-log /dev/nvme0n1 -i 0xc5 -l 0x4 -o 0x4
```
#### Trace for nvme driver
```bash
nvme_setup_admin_cmd:  cmdid=21, flags=0x0, meta=0x0, cmd=(nvme_admin_get_log_page cdw10=c5 00 00 00 00 00 00 00 04 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00)
nvme_complete_rq: cmdid=21, qid=19, res=0, retries=0, flags=0x0, status=16385
```
Trace shows that the command whose cdw10 that is an offset(CDW12) has 0x4 as the offset like the trace above.
I think the nvme-cli has passed the command to the device thorugh the driver properly.

Which nvme-cli version are you using?

blguerke, 2018-04-17:
Thanks Minwoo Im for the support.
I'm using Linux 4.16.2 and nvme-cli version 1.5.148.gea19
Haven't tried the nvme driver trace tool, but when I run the command I just don't get the offset specified.
Ex: 
# nvme get-log /dev/nvme0n1 -i 0xc5 -l 0x8
10 00 00 00 95 11 00 00 --> now I want to get 95 11 00 00, so i'd pass a offset 0x4...
# nvme get-log /dev/nvme0n1 -i 0xc5 -l 0x4 -o 0x4
10 00 00 00  --> not getting 95 11 00 00 which is what I expected

Is this repeatable on your end?

blguerke, 2018-04-23:
Yeah, that may be the reason. I was hoping to avoid outputting this to a bin and post processing the data for the values needed, but that may be all that's available. Thanks for the trace capture help. Here's what mine shows:
`nvme_setup_admin_cmd:  cmdid=9, flags=0x0, meta=0x0, cmd=(nvme_admin_get_log_page cdw10=c5 00 00 00 00 00 00 00 04 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00)`
`nvme_complete_rq: cmdid=9, qid=8, res=0, retries=0, flags=0x0, status=0`

minwooim, 2018-04-24:
Then I guess the device you have may not support the offset feature in get-log. :(

closed 2022-01-19 by igaw