linux-nvme/nvme-cli #338: Non-zero return value for "show-regs" command when it succeeds

Shadiasd, 2018-04-12:
Hi,

I noticed that the return value for `nvme show-regs` command is non-zero (234) when it succeeds. All the other nvme-cli commands that I used so far return a zero value if successful.

- **Steps required to generate the issue:**
`sudo nvme show-regs /dev/nvme0`
`echo $?`

- **nvme-cli version:** 1.5

Please note that the above issue does not exist in nvme-cli version 1.3 and the return value is zero if the command is successful.

It would be great if you can fix this issue.

Thank you,
Shadi

minwooim, 2018-04-12:
Hi Shadi,
I have submitted a PR #339 to fix this issue.

Shadi and Keith,
Would you please look into this patch?

Shadiasd, 2018-04-16:
Thank you for the fix! :)