linux-nvme/nvme-cli #344: Unable to redirect the output of fw-log -o json command to a file - nvme-cli version 1.3

Shadiasd, 2018-04-16:
Hi, 

I noticed that the output of `fw-log` command in json format cannot be redirected to a file. This is an issue with the nvme-cli version 1.3 and does not exist in versions 1.4 or 1.5. 

**Steps to create the issue:**

- **nvme-cli version**: 1.3 (relesed version)
- **nvme-cli commnd**: `sudo ./nvme fw-log /dev/nvme0 -o json 1> out.json 2> stderr`
- The problem seems to be with stdout, since if I intentionally induce an error in the command, I can see the errro being redirected to stderr

****Note**: I did the same redirection with `smart-log -o json` command and `fw-log` command without json format, and in both cases it successfully stores the output to the file:
`sudo ./nvme smart-log /dev/nvme0 -o json 1> out.json 2> stderr`
`sudo ./nvme fw-log /dev/nvme0  1> out.txt 2> stderr`

Unfortunately, for the setting in which we use nvme-cli, there is no option to use any version older than the 1.3 release. Hence, I need to find a fix for this issue in version 1.3. However, looking at the code and commit history I cannot find out what is causing this issue, and which commit(s) fixed it.

It will be great if you can suggest a fix or point me to the commit that has fixed this issue.

Any help is greatly appreciated!
Shadi

Shadiasd, 2018-04-17:
A bit more information that might help debugging this issue:

I found out that I can successfully store the output to a file (json.txt) using "script" command as follows: 
`sudo script -q -c './nvme fw-log /dev/nvme1 -o json' /dev/null > json.txt`

This command seems to be a workaround to capture the output of the programs where the output is written to /dev/tty. Although this solution works for nvme-cli, looking at nvme-cli code I do not think that the output is actually being written to /dev/tty. Additionally, I do not have the option to use "script" to overcome this issue either. Hence, I need to apply a fix to nvme-cli code that let me redirect the output of `fw-log -o json` command to a file.

Thank you,
Shadi

minwooim, 2018-04-17:
Hi Shadiasd,

The command below has been fixed by commit ac78cef4d3 by Keith for #182 .
nvme-cli v1.3 has a bug which double-freed the json object.
Please refer this commit(ac78cef4d3) and PR #182. 

Thanks,

keithbusch, 2018-04-17:
Is the request to provide bug fixes on tagged versions? I never considered maintaining a stable branch.

Shadiasd, 2018-04-17:
Hi,

@minwooim, thank you for your help! It does fix the redirection problem! After applying the fix in commit [ac78cef](https://github.com/linux-nvme/nvme-cli/commit/ac78cef4d3cf3b03afee6d4eb5ed5955385ae606), I can successfully redirect the stdout to a file.

@keithbusch, no I did not mean to ask that. I was just hopping that maybe some one can point me to the right commit that fixed this issue. 

None: I did do the search myself prior to submitting this issue and did come across a few commits including the one that minwooim pointed out, but couldn't figure out which one had fixed the redirection problem as none of them seem to be addressing this issue directly, although ac78cef does actually fix it!  

Anyway, thank you for your help :)

closed 2018-04-17 by Shadiasd