linux-nvme/nvme-cli #346: namespace operation on optane 900P?

fengggli, 2018-04-19:
#### Description:
I have a intel 900P installed in a Ubuntu 16.04 machine (with nvme 1.0 kernel module loaded by default)
Some of the system information: (it's recognized as 2700... not 900P, I am not sure it's critical or not..)
```
lifeng@1wk300(x-pheonix:2):~/Workspace/x-phoenix/build$lspci |tail -1

04:00.0 Non-Volatile memory controller: Intel Corporation Device 2700

lifeng@1wk300(x-pheonix:2):~/Workspace/x-phoenix/build$uname -a

Linux 1wk300 4.13.0-38-generic #43~16.04.1-Ubuntu SMP Wed Mar 14 17:48:43 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux
```
I saw the output of the fdisk saying it has 512 byte sector size and i/o size:
```
Disk /dev/nvme0n1: 260.9 GiB, 280065171456 bytes, 547002288 sectors Units: sectors of 1 * 512 = 512 bytes Sector size (logical/physical): 512 bytes / 512 bytes I/O size (minimum/optimal): 512 bytes / 512 bytes 
```
#### My questions:
1. Is there any way to reformat the nvme to 4k sector size? I am assuming I will achieving better performance instead of using 512 block size.

2. I have tried the newest release of nvme-format(from the nvme-cli) .  using the default configurations(nvme-format + device name) gives the same blocks/sector size. I also tried to delete the default namespaces(nvme detach-ns + nvme delete-ns) and create a new namespace of 4k size (create-ns), but it failed .. some of the output(the operation code sent by nvme-cli somehow is not correct..):
```
lifeng@1wk300(x-pheonix:2):~/Workspace/x-phoenix/build$sudo nvme detach-ns /dev/nvme0 --namespace-id=0x1

NVMe Status:INVALID_OPCODE(4001)

lifeng@1wk300(x-pheonix:2):~/Workspace/x-phoenix/build$sudo nvme list-ns /dev/nvme0 --namespace-id=0x1

[   0]:0x209a97b0

[   2]:0x209a97b0

[   4]:0x209a97b0

[  32]:0x2090000

lifeng@1wk300(x-pheonix:2):~/Workspace/x-phoenix/build$sudo nvme list-ns /dev/nvme0

NVMe Status:INVALID_NS(400b) NSID:0

lifeng@1wk300(x-pheonix:2):~/Workspace/x-phoenix/build$sudo nvme id-ns /dev/nvme0 --namespace-id=0x1

NVME Identify Namespace 1:

nsze    : 0x209a97b0

ncap    : 0x209a97b0

nuse    : 0x209a97b0

nsfeat  : 0

nlbaf   : 0

flbas   : 0

mc      : 0

dpc     : 0

dps     : 0

nmic    : 0

rescap  : 0

fpi     : 0

nawun   : 0

nawupf  : 0

nacwu   : 0

nabsn   : 0

nabo    : 0

nabspf  : 0

noiob   : 0

nvmcap  : 0

nguid   : 00000000000000000000000000000000

eui64   : 0000000000000000

lbaf  0 : ms:0   lbads:9  rp:0x2 (in use)
```

Any comments or suggestions are appreciated!
Thanks

keithbusch, 2018-04-20:
If this did support that format, there should have been an 'lbaf' with an 'lbads:12', and then you can select that with an "nvme format /dev/nvmeX -l <lbaf>'. I'm not sure if this one supports 4k or not. The optanes I have do support that, but they're slightly different than the 900P.

In any case, if you align IO and partitions to 4k, you should get the same performance out of this as if it were formatted to 4k physical sectors.

fengggli, 2018-04-20:
hi Keith, thanks for the explanation
1. "If this did support that format, there should have been an 'lbaf' with an 'lbads:12"
I know there is only one lbaf value in my id-ns output: ```lbaf  0 : ms:0   lbads:9  rp:0x2 (in use)```
**How can I know what are the other supported formats of my device?(other than format 0)**
the man page says  
  *LBA Format: This field specifies the LBA format to apply to the NVM media. This corresponds to the LBA formats indicated in the Identify Namespace command.*
2. Could you explain more on how to align I/O to 4k?
Are there specific tools I can use to change the I/O size?(I won't make any partitions in the nvme0n1)
At least this didn't work -_-:
```
echo 4096 > /sys/block/nvme0n1/queue/minimum_io_size
```

Thanks

rolandqi, 2018-05-29:
@fengggli if your id-ns only have one lbaf listed, that's means it only support one format. Maybe you should contact your SSD FW vendor for more format support.

darkbasic, 2018-05-29:
> Maybe you should contact your SSD FW vendor for more format support.

He did: https://communities.intel.com/thread/124672

rolandqi, 2018-05-29:
I think maybe some enterprise Optane products support extra format.

fengggli, 2018-05-30:
@darkbasic Thanks for pointing out. 
@rolandqi  you are right.  this device only supports one format..

closed 2018-05-30 by fengggli