linux-nvme/nvme-cli #350: If nvme devices is mounted, and user doing disconnect same device then cli should throw error.

sppavilion, 2018-05-08:
> nvme-cli version - 1.5

**Steps followed :** 

- Connect nvme device.
- Create any filesystem(ext4/xfs) on nvme device
- Mount device.
- Disconnect nvme device using 'nvme disconnect -d nvmexn1'

**Observation**

1. Block layer doesn't understand the device is mounted or not. Hence disconnect device operation/process get into hung state.

**Expected output**

1. If user is doing forceful disconnect nvme device which is mounted, then nvme cli should give proper error. [Instead of executing disconnect]

--------------------------------------------------------------------------------------------------------------
**Shubham Pardeshi**

keithbusch, 2018-05-08:
I'm not sure it's this tool's job to know if you've mounted a block device or not. That's really not even practical to do at this layer.

hreinecke, 2022-01-16:
Indeed. nvme-cli is manipulating the drive itself, with no interaction to the kernel details. So you can do arbitrary things, some of which are known to be harmful when the kernel is accessing the drive.

"Don't do it, then!"

closed 2022-01-16 by hreinecke