linux-nvme/nvme-cli #351: gcc 4.8 support

tim-oleksii, 2018-05-15:
Any plans to support gcc 4.8? It used to support it last year, but with https://github.com/linux-nvme/nvme-cli/commit/bd1f2f61541d6ecbef1ea17e6c938a62848daa00 build breaks are encountered.

Here is the build log
[gcc_4_8_build.log](https://github.com/linux-nvme/nvme-cli/files/2004239/gcc_4_8_build.log)

keithbusch, 2018-05-15:
this is the first it's come to my attention. we never had a plan to break on gcc 4.8 in the first place. will look into it.

tim-oleksii, 2018-05-15:
I've got distros up and running with gcc 4.8, you can send for testing.
It's good to place forward declaration of endurance log or include "linux/nvme.h", however, I dunno how do you feel about it? If it's good, I can resolve it.

keithbusch, 2018-05-15:
diff --git a/nvme-print.h b/nvme-print.h
index 3372b28..42f56b1 100644
--- a/nvme-print.h
+++ b/nvme-print.h
@@ -5,6 +5,8 @@
 #include "json.h"
 #include <inttypes.h>

+#include "linux/nvme.h"
+
 enum {
        TERSE = 0x1u,   // only show a few useful fields
        HUMAN = 0x2u,   // interpret some values for humans

tim-oleksii, 2018-05-15:
The more I test, the more it comes to my mind that my environment is broken. Cloning nvme-cli and building it with gcc 4.8.2 is OK and fine, sorry for disturbance.

closed 2018-05-15 by tim-oleksii