linux-nvme/nvme-cli #354: JSON generator is locale-dependent

kappa, 2018-05-21:
Running LC_ALL=ru_RU.UTF-8 /usr/sbin/nvme id-ctrl $device -o json may produce '"tnvmcap" : 0,000000,' because the decimal separator is comma in that locale.

JSON should always be generated in "C" locale.

I would vote to remove setlocate(LC_ALL, ""), which means "use locale env vars", from main() altogether but I may not know why it is there in the first place.

keithbusch, 2018-06-15:
Thanks for the notice. The reason it's there is because there are some 128-bit decimal numbers that are rather unfriendly to read without the locale. It certainly doesn't belong in the json output, though.

igaw, 2022-01-19:
No progress in this bug report for a long time. If you think it's still relevant please reopen.

closed 2022-01-19 by igaw

piotr-fraczek, 2022-09-29:
Version 2.1.2 still have problem:
```
"tnvmcap":"0,000000",
"unvmcap":"0,000000",
```

igaw, 2022-09-29:
We just fixed the 16 bytes values problems in the tree. It's not yet released yet.

b509ccb39a8c ("utils: Fix uint128_t usage")