linux-nvme/nvme-cli #358: nvme --list --output-format=json Non JSON output

paultop6, 2018-05-25:
When running command nvme --list --output-format=json and there are no nvme devices detected, the output is a standard string of value "No NVMe devices detected." instead of standard JSON, like the following:
`{
  "Devices" : [ ]
}`

keithbusch, 2018-05-25:
That's to stderr, though. That shouldn't get picked up by anything trying to consume json output, right?

paultop6, 2018-05-25:
Double checked, its coming out stdout, and return code set to 0 (checked $?)

keithbusch, 2018-05-25:
Hm? This is the line of code that's printing, right?

`fprintf(stderr, "no NVMe device(s) detected.\n");`

paultop6, 2018-05-25:
Feel like a fool, was using an older version.  Tested the same scenario with 1.5 and no issues.  Will close issue now.

closed 2018-05-25 by paultop6