linux-nvme/nvme-cli #360: Why allocate extra block size in submit_io

rolandqi, 2018-05-30:
In the submit_io() in nvme.c, the buffer_size is allocated 1 more than the actual block_count. Is there any special ends of this behavior? Is this to prevent the drive using metadata format step outside the data memory?

`	buffer_size = (cfg.block_count + 1) * phys_sector_size;`

tim-oleksii, 2018-05-30:
The value is zero based, it's necessary to add one more block to get actual count.

https://github.com/linux-nvme/nvme-cli/blob/d94fda25db81ba5b43b18c05dadff489d50ea0d2/nvme.c#L3514

rolandqi, 2018-05-31:
Sorry I miss that...  But if so, How can we secure the data integration if the drive is formatted with metadata and would be transfer as a part of the LBA? Because it haven't allocated extra memory space for the metadata.

keithbusch, 2018-06-04:
You'll have to override with the default with the '--data-size=<size-in-bytes>' option.

rolandqi, 2018-06-06:
So it's basically not the tool's duty to secure this process if the `--data-size=<size-in-bytes>` not provided meaningfully. Thanks for your input!

closed 2018-06-06 by rolandqi

keithbusch, 2018-06-06:
Yeah, we could build more smarts into the tool to discover parameters to assist a user, but I wanted to keep this tool "dumb", with a 1:1 command line:nvme command to keep noise down if you were observing behavior to a specific command on a protocol analyzer or with nvme event tracer. If you want to do something clever, I tried to make this friendly to scripting and piping.