linux-nvme/nvme-cli #367: NVME-VERSION-GEN: Permission denied

herzlich, 2018-06-11:
env: RHEL 7.4, IBM Power Server

fails on 1.4 and latest 11-Jun-2018 (didn't check on older releases)

ls -l NVME-VERSION-GEN
-rw-r--r--. 1 root root 761 Jun 11 03:00 NVME-VERSION-GEN

should be 755 for make install to run.

Symptoms:
[root@blancapk nvme-cli-master-latest]# make install
make: execvp: ./NVME-VERSION-GEN: Permission denied
cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -DNVME_VERSION='""' -c argconfig.c
cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -DNVME_VERSION='""' -c suffix.c
cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 
...

tim-oleksii, 2018-06-11:
Did you clone it from upstream or copied from somewhere else?

> [tim-oleksii@rediska tests]$ git clone https://github.com/linux-nvme/nvme-cli.git && cd nvme-cli && ls -la | grep NVME
> Cloning into 'nvme-cli'...
> remote: Counting objects: 4937, done.
> remote: Compressing objects: 100% (10/10), done.
> remote: Total 4937 (delta 6), reused 12 (delta 5), pack-reused 4922
> Receiving objects: 100% (4937/4937), 2.28 MiB | 1.27 MiB/s, done.
> Resolving deltas: 100% (3703/3703), done.
> 47:-rwxr-xr-x 1 tim-oleksii users    761 Jun 11 18:34 NVME-VERSION-GEN

herzlich, 2018-06-11:
User error: Issue caused by unzipping on different system and moving folder to destination.  Default file mask applied to individual files rather than settings inside zip file.

closed 2018-06-11 by herzlich