linux-nvme/nvme-cli #371: nvme-cli: smart-log format is inconsistent

tim-oleksii, 2018-06-12:
> [tim-oleksii@rediska nvme-cli]$ sudo ./nvme smart-log /dev/nvme0 -o normal
> Smart Log for NVME device:nvme0 namespace-id:ffffffff
> critical_warning                    : 0
> temperature                         : 39 C
> available_spare                     : 100%
> available_spare_threshold           : 10%
> percentage_used                     : 0%
> data_units_read                     : 28
> data_units_written                  : 0
> host_read_commands                  : 487
> host_write_commands                 : 0
> controller_busy_time                : 0
> power_cycles                        : 13
> power_on_hours                      : 189
> unsafe_shutdowns                    : 6
> media_errors                        : 0
> num_err_log_entries                 : 0
> Warning Temperature Time            : 0
> Critical Composite Temperature Time : 0
> Temperature Sensor 1                : 39 C
> Temperature Sensor 2                : 77 C
> Thermal Management T1 Trans Count   : 0
> Thermal Management T2 Trans Count   : 0
> Thermal Management T1 Total Time    : 0
> Thermal Management T2 Total Time    : 0
>

minwooim, 2018-06-12:
Hi Tim,
You mean the first line of the output?
```Smart Log for NVME device:nvme0 namespace-id:ffffffff```

keithbusch, 2018-06-12:
Inconsistent in that sometimes it's a underscore_separated_field and other times is a Expanded Field Name?

tim-oleksii, 2018-06-12:
Ah, sorry, I didn't think that it can be incosistent in two or more ways, @keithbusch mentioned the option I thought about.

ScottyBauer, 2018-06-12:
What way would you prefer? All underscores or all spaces?

tim-oleksii, 2018-06-12:
I was wondering whether consistency should be held across one command. Should we keep it?
There is no easy answer to prefer spaces against underscores as vendor plugins have already handled it differently. Enforcing it right now for everyone is bad option as some outputs are considered to be close to some vendor specific analogues of nvme-cli. 

On the other hand, if the command sends output with underscores, someone would expect consistent behaviour. I'll go with keeping old behaviour, despite the personal taste.

What do you think? Should it be considered?

keithbusch, 2019-03-18:
Just going to leave it as-is. Some external test tooling tends to freak out when the output changes, so I have to more careful on changing user visible output. :(

closed 2019-03-18 by keithbusch