linux-nvme/nvme-cli #372: nvme-cli: -h option doesn't work for telemetry-log

tim-oleksii, 2018-06-12:
Recently tried to get telemetry from my device, unfortunately, it failed with '-h'.

```
[tim-oleksii@rediska tim_nvme_cli]$ sudo ./nvme telemetry-log /dev/nvme0 -h 1 -o /tmp/out.bin
Usage: nvme telemetry-log <device> [OPTIONS]

Retrieve telemetry log and write to binary file

Options:
  [  --output-file=<FILE>, -o <FILE> ]  --- File name to save raw binary,
                                            includes header
  [  --host-generate=<NUM>, -h <NUM> ]  --- Have the host tell the controller
                                            to generate the report
  [  --controller-init, -c ]            --- Gather report generated by the
                                            controller.
  [  --data-area=<NUM>, -d <NUM> ]      --- Pick which telemetry data area to
                                            report. Default is all. Valid
                                            options are 1, 2, 3.

```

However, the long option works.

```
[tim-oleksii@rediska tim_nvme_cli]$ sudo ./nvme telemetry-log /dev/nvme0 --host-generate=1 -o /tmp/out.bin
[tim-oleksii@rediska tim_nvme_cli]$ echo $?
0

```

'-h' is treated like '--help'  by argconfig.c
https://github.com/linux-nvme/nvme-cli/blob/2e65d5ef3b994796c33dfd316bf87f6932d125c5/argconfig.c#L204

What do you think about --generate-host or -g?

PS: btw, this seems the reason why id-ctrl works only with -H (i skipped it as it's 'human-readable', didn't want to mess with double meaning).

keithbusch, 2018-06-12:
I don't really like the "-H" options either. That really should have been a --verbose/-v option.

But anyway, yeah, that's a good catch. I'll stare at this one for a bit and see if there's a good solution.

keithbusch, 2018-06-15:
Fixed now.

closed 2018-06-15 by keithbusch