linux-nvme/nvme-cli #373: ns-descs INVALID_FIELD(4002) on PLEXTOR PX-1TM9PeG

asava, 2018-06-13:
Hello,

Nvme v1.5 with PLEXTOR PX-1TM9PeG (1TB M9Pe) drive, reports the following error:

`$ sudo nvme ns-descs /dev/nvme0 -n 1
NVMe Status:INVALID_FIELD(4002) NSID:1`

Namespace 1 is defined and valid:

`$ sudo nvme list 
Node             SN                   Model                                    Namespace Usage                      Format           FW Rev  
---------------- -------------------- ---------------------------------------- --------- -------------------------- ---------------- --------
/dev/nvme0n1     ............         PLEXTOR PX-1TM9PeG                       1           1,02  TB /   1,02  TB      4 KiB +  0 B   1.04    `

`$ sudo nvme id-ns /dev/nvme0 -n 1
NVME Identify Namespace 1:
nsze    : 0xee77a56
ncap    : 0xee77a56
nuse    : 0xee77a56
nsfeat  : 0
nlbaf   : 1
flbas   : 0x11
mc      : 0
dpc     : 0
dps     : 0
nmic    : 0
rescap  : 0
fpi     : 0
nawun   : 0
nawupf  : 0
nacwu   : 0
nabsn   : 0
nabo    : 0
nabspf  : 0
noiob   : 0
nvmcap  : 1024209543168
nguid   : 00000000000000000000000000000000
eui64   : 0023035630030fd9
lbaf  0 : ms:0   lbads:9  rp:0x1 
lbaf  1 : ms:0   lbads:12 rp:0 (in use)`

keithbusch, 2018-06-13:
Please see NVMe Specification Figure 106 for the details on how this command operates. The pertinent part is:

>   A list of 1024 namespace IDs is returned containing active NSIDs in increasing order that are greater than the value specified in the Namespace Identifier (CDW1.NSID) field of the command.

So if your highest NSID is 1, then setting NSID to 1 in the ns-descs command is certainly invalid.

keithbusch, 2018-06-13:
Err, scratch that, I was confusing this with the "ns-list" command... This probably means your device doesn't support the Namespace Descriptor Identify mode.

asava, 2018-06-13:
Thank you for your reply, I understand now. I thought it might be an incompatibility that could be solved. If that was the case, I could provide debugging information. 
But if the drive doesn't support that not much to do.

closed 2022-01-19 by igaw