linux-nvme/nvme-cli #374: nvme-cli: log page 80h is not supported

tim-oleksii, 2018-06-13:
I browsed the code recently, noticed that 80h is not supported, right now. Could some one please why it's so? I think that messing with RAE is not right idea, but getting data without clearing async status seems OK.

Sorry, if I missed any discussion prior to this one, browsing through issues had shown nothing. infradead is also silent.

hreinecke, 2019-02-12:
Log page 80h is unaffected by RAE, and every read on the log page clears that particular notification.
Which means that we would be interfering with any other software trying to work with reservations.
So I don't think we'll be implementing it.

tim-oleksii, 2019-02-13:
This case came to my mind long ago, however, I forgot to close the issue. Thank you for confirming it.

closed 2019-02-13 by tim-oleksii