linux-nvme/nvme-cli #376: NVMe rubsystem-reset support and behaviour

narasimhan-v, 2018-06-18:
I have an nvme subsystem which does not support subsystem-reset.
I suppose the NSSRC bit is the one which gives this information.

```
# nvme show-regs /dev/nvme0 -H | grep -i subsystem
	NVM Subsystem Reset Supported  (NSSRS): Yes
	NVM Subsystem Reset Occurred (NSSRO): No
	NVM Subsystem Reset Control (NSSRC): 0
```

But the `nvme subsystem-reset` command runs fine.
Isn't the command supposed to throw an error ? Or what is the expected behaviour ?

narasimhan-v, 2018-06-18:
```
# nvme version
nvme version 1.2
```

narasimhan-v, 2018-06-18:
But should that write check for the NSSRC bit ?

keithbusch, 2018-06-18:
NSSRC is not a bit, and is defined by spec to always return 0 when read.

keithbusch, 2019-01-18:
The kernel side implementation is a bit incomplete. Subsystem reset will take the link down, many ports don't report the link event for the driver to react to, so we end up with an unconfigured device.

closed 2022-01-19 by igaw