linux-nvme/nvme-cli #386: [Centos 6.9] : nvme cli make command failed - "nvme-lightnvm.c:244: error: dereferencing pointer ‘id.47’ does break strict-aliasing rules"

sppavilion, 2018-07-05:
[root@localhost nvme-cli]# cat /etc/centos-release 
**CentOS release 6.9 (Final)**
[root@localhost nvme-cli]# 
[root@localhost nvme-cli]# uname -r 
**2.6.32-696.el6.x86_64**
[root@localhost nvme-cli]# 

[root@localhost nvme-cli]# make
cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -DNVME_VERSION='"1.5.232.g13e20"' -c argconfig.c
cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -DNVME_VERSION='"1.5.232.g13e20"' -c suffix.c
cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -DNVME_VERSION='"1.5.232.g13e20"' -c parser.c
cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -DNVME_VERSION='"1.5.232.g13e20"' -c nvme-print.c
cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -DNVME_VERSION='"1.5.232.g13e20"' -c nvme-ioctl.c
cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -DNVME_VERSION='"1.5.232.g13e20"' -c nvme-lightnvm.c
cc1: warnings being treated as errors
nvme-lightnvm.c: In function ‘lnvm_do_id_ns’:
nvme-lightnvm.c:244: error: dereferencing pointer ‘id.47’ does break strict-aliasing rules
nvme-lightnvm.c:245: error: dereferencing pointer ‘id.47’ does break strict-aliasing rules
nvme-lightnvm.c:247: error: dereferencing pointer ‘id.47’ does break strict-aliasing rules
nvme-lightnvm.c:249: error: dereferencing pointer ‘id.47’ does break strict-aliasing rules
nvme-lightnvm.c:254: error: dereferencing pointer ‘id.47’ does break strict-aliasing rules
nvme-lightnvm.c:256: error: dereferencing pointer ‘id.47’ does break strict-aliasing rules
nvme-lightnvm.c:258: error: dereferencing pointer ‘id.47’ does break strict-aliasing rules
nvme-lightnvm.c:260: error: dereferencing pointer ‘id.47’ does break strict-aliasing rules
nvme-lightnvm.c:265: error: dereferencing pointer ‘id.47’ does break strict-aliasing rules
nvme-lightnvm.c:266: error: dereferencing pointer ‘id.47’ does break strict-aliasing rules
nvme-lightnvm.c:267: error: dereferencing pointer ‘id.47’ does break strict-aliasing rules
nvme-lightnvm.c:268: error: dereferencing pointer ‘id.47’ does break strict-aliasing rules
nvme-lightnvm.c:269: error: dereferencing pointer ‘id.47’ does break strict-aliasing rules
nvme-lightnvm.c:270: error: dereferencing pointer ‘id.47’ does break strict-aliasing rules
nvme-lightnvm.c:271: error: dereferencing pointer ‘id.47’ does break strict-aliasing rules
nvme-lightnvm.c:272: error: dereferencing pointer ‘id.47’ does break strict-aliasing rules
nvme-lightnvm.c:273: error: dereferencing pointer ‘id.47’ does break strict-aliasing rules
nvme-lightnvm.c:274: error: dereferencing pointer ‘id.47’ does break strict-aliasing rules
nvme-lightnvm.c:275: error: dereferencing pointer ‘id.47’ does break strict-aliasing rules
nvme-lightnvm.c:276: error: dereferencing pointer ‘id.47’ does break strict-aliasing rules
nvme-lightnvm.c:277: error: dereferencing pointer ‘id.47’ does break strict-aliasing rules
nvme-lightnvm.c:278: error: dereferencing pointer ‘id.47’ does break strict-aliasing rules
nvme-lightnvm.c:311: error: dereferencing pointer ‘id.47’ does break strict-aliasing rules
nvme-lightnvm.c:371: error: dereferencing pointer ‘id.47’ does break strict-aliasing rules
nvme-lightnvm.c:424: note: initialized from here
make: *** [nvme-lightnvm.o] Error 1
[root@localhost nvme-cli]#

[root@localhost nvme-cli]# git log
commit 13e20576cca6ea655e2f9ffed1f8680e369d5bdf
Author: Minwoo Im <minwoo.im.dev@gmail.com>
Date:   Tue Jul 3 22:18:59 2018 +0900

    nvme-cli: adjust memset size for data buffer in submit_io
    
    Adjust size of memset() for the data buffer to its own size.
    
    Signed-off-by: Minwoo Im <minwoo.im.dev@gmail.com>

keithbusch, 2018-07-05:
Huh, so this is saying the struct doesn't overlay with what it's told to point to. Everything is supposed to pack such that it does overlay, but your compiler is having some problems with that. Doesn't seem to be a problem on newer gcc's, but we'd like to avoid compiler dependencies whenever possible. I don't know the lighnvm stuff very well at the moment, but I'll try to get some time to get back on this one.

keithbusch, 2018-07-17:
afraid i may need to reach out to the open channel folks on how this is supposed to look. I can make the errors go away, but i think some of these structures may have the wrong definition from the beginning, and i'm not sure what they're supposed to look like.

keithbusch, 2018-07-17:
the patch linked here should fix the compiler errors and should be equivalent to what was there before, but bypasses the compiler checks by casting everything to void in the interim.

igaw, 2022-01-19:
No progress in this bug report for a long time. If you think it's still relevant please reopen.

closed 2022-01-19 by igaw

keithbusch, 2022-01-19:
i think all lightnvm should be removed from the tool. the kernel removed the module several releases ago, so it wouldn't work anyway.