linux-nvme/nvme-cli #388: nvme list no longer includes PCI ID for each device in the output

excelero-joe, 2018-07-12:
I recall in a previous release, or perhaps a modified release from a 3rd party, the `nvme list` command used to provide the PCI ID of each NVMe.  This is an extremely valuable correlation to make and would be nice to have in the tool.

Additionally, if the slot ID in `/sys/bus/pci/slots/*` could be printed as well, this would really round out the tool's ability to provide critical PCIe device info.

keithbusch, 2018-07-12:
By PCI ID, are you referring to the VID:DID? You can get that with:

  `cat /sys/class/nvme/nvmeX/device/{vendor,device}`

The whole /sys/bus/pci/slots hierarchy needs some serious work to be more usable. There's no direct way to connect /dev/nvme0 to /sys/bus/pci/slots/<slot#> without scanning all sorts ... At the very least, I think the PCI subsystem ought to have a symlink from the NVMe parent bridge to the slot directory if this. 

In any case, this doesn't really fit in with the fabrics stuff, but I suppose we can add PCI-centric if it really helps

excelero-joe, 2018-07-12:
The PCI ID to which I am referring is the one listed in `lspci -vvv | grep Non-V` in the first column of output.  I often need this value to correlate with the logical PCIe slot ID for things like powering down the drive slot for pro-active media replacement.  Having both items displayed in this one tool could be very helpful indeed.  Thanks!

keithbusch, 2018-07-12:
Oh, that's the PCI B:D.f, not the PCI ID. Get that with: `readlink -f /sys/class/nvme/nvmeX`

excelero-joe, 2018-07-12:
Yes!  This is the exact detail I would like to see the tool report in its `nvme list` output.

keithbusch, 2018-07-12:
That's one line of bash, but a pain in C ... I'll look into it.

freezer5, 2019-07-17:
Recalling this old issue.
The "nvme list" command doesn't really suit this purpose, so I propose to add a new command called "pci-list" which will show the PCI B:D.f address alongside the NVMe namespace information.
Please see https://github.com/linux-nvme/nvme-cli/commit/bff339adea25742ff071cc01ada76072fe1c7b10 .

hreinecke, 2022-01-16:
Fixed now. Closing.

closed 2022-01-16 by hreinecke