linux-nvme/nvme-cli #391: Additional SMART Log produces `NVMe Status:INVALID_LOG_PAGE(2109)`

iwittkau, 2018-07-23:
```

keithbusch, 2018-07-23:
That's an optional, vendor specific log page. Most drives don't support it.

iwittkau, 2018-07-23:
Wow, that was quick a quick answer, was still editing ... so this is not supported by my SSD?

keithbusch, 2018-07-23:
Yeah, sounds like it isn't supported by your vendor.

keithbusch, 2018-07-23:
It's an invalid log page. What more does a user want to see?

keithbusch, 2018-07-23:
Note, a more recent version will use the error string defined in the specification in addition to the name of the status field, so it would look something like this:

"NVMe Status:INVALID_LOG_PAGE: The log page indicated is invalid. This error condition is also returned if a reserved log page is requested(2109)"

iwittkau, 2018-07-23:
To me it just looked like `nvme` was broken or the drive support was inclomplete. `INVALID_LOG_PAGE` still seems like a very general error to me.  
If this is all based on a specification than it makes sense, I guess. Anyway, thanks for the great tool!

closed 2018-07-23 by iwittkau