linux-nvme/nvme-cli #400: nvme io-passthru - linux kernel error

celebum, 2018-08-09:
Hello

I was try to run a nvme io-passthru command on my nvme devices on Ubuntu 18.04.1.

For read everything is looks good
```
nvme io-passthru -r -l 512 -n 1 -i /dev/zero /dev/nvme0n1
```
Unfortunately for write the kernel returns strange message:
```
nvme io-passthru -w -l 512 -n 1 -i /dev/zero /dev/nvme0n1
```
[image]

Do anybody can help with that?

keithbusch, 2018-08-09:
The 'io-passthru' is for you to define the command dwords, which includes the opcode. It doesn't look like you're setting the opcode, though, so you're sending an opcode 0 in both cases which is a 'flush' command. You need to set '-o 0x1' of write, and '-o 0x2' for read.

Not sure why you're getting command effects on an io opcode 0, though, and it should be the same for both the '-r' and the '-w' cases.

keithbusch, 2018-08-09:
BTW, the 'io-passthru' is provided to future proof the tool to new spec changes and devices for opcodes we don't know about yet. For every opcode that is defined, the tool provides more convenient sub-commands so you don't need to individually define dwords. There are 'nvme write' and 'nvme read' commands for this sort of thing.

closed 2022-01-19 by igaw