linux-nvme/nvme-cli #401: Docs for /sys/kernel/config/nvmet/hosts ?

CrucialDrew, 2018-08-11:
I realize this isn't specific to nvme-cli, but I'm hoping to get some assistance.

I haven't had much luck finding any good explanation for how to authorize specific hosts to connect to an nvmet subsystem.

I see that there is:

/sys/kernel/config/nvmet/hosts

And 

/sys/kernel/config/nvmet/subsystems/*/allowed_hosts

So I'm guessing I create something in /sys/kernel/config/nvmet/hosts and symlink it in the subsystem.

Would you mind shedding some light on how this works, or pointing me to the appropriate docs? It would be nice to not have to dig through the source (but I will :-P)

keithbusch, 2018-08-13:
While we work together, this project is not affiliated with nvmet. I think the documentation here would be more useful: http://git.infradead.org/users/hch/nvmetcli.git/blob/8b9ce6be23788c80ff692ccb9d985cf97afc936e:/Documentation/nvmetcli.txt

CrucialDrew, 2018-08-13:
Thank you, I was able to get it figured out from there.

closed 2018-08-13 by CrucialDrew