linux-nvme/nvme-cli #404: Know when a device is "internal"

hughsie, 2018-08-16:
I'm planning to use nvme-cli from fwupd for updating firmware. I'm interested to know if anything in id-ctrl indicates if the NMVe device indicates the device is "internal" to the machine or if it is connected using a external bridge. I guess I could walk the PCI tree, but I wondered if we could use something in the existing json. Thanks!

keithbusch, 2018-08-16:
Not even sure how walking the PCI tree would help. The only that could tell you is if your device is connected through a bridge or direct connect to a root port. It wouldn't tell you if the bridge is internal or external.

A direct root port connect:

```
# readlink -f /sys/class/nvme/nvme2
/sys/devices/pci0000:00/0000:00:01.1/0000:05:00.0/nvme/nvme2
```

A switch/bridge connect:

```
# readlink -f /sys/class/nvme/nvme0
/sys/devices/pci0000:00/0000:00:01.0/0000:01:00.0/0000:02:01.0/0000:03:00.0/nvme/nvme0
```

hughsie, 2018-08-16:
Wow, thanks for the quick answer. I wondered if looking at the parent to see if any other devices are common would work.

hughsie, 2018-08-23:
I've added some heuristics based on the topology, no worries.

closed 2018-08-23 by hughsie