linux-nvme/nvme-cli #406: How to use nvme security-send / security-recv command

seanmaster, 2018-08-28:
Hi All, everybody,

I want-to know how to use security-send/recv command? I don't know how to issue it. Can I get any example about this command? Very Thanks.

keithbusch, 2018-08-28:
The NVMe specification only provides the commands to send and receive security related data. The NVMe specification does not define the structure of that data, leaving it to the controller to choose and implement a protocol. As there is no standard security protocol to decode or create those data packets, the security send/receive commands are quite simple in nvme-cli. You need to supply the data streams through another program.

seanmaster, 2018-10-17:
Thanks your reply.

closed 2018-10-17 by seanmaster

rvanlaar, 2019-09-24:
This is a bit of a weird answer. 
The documentation is very bare bones. For example, what do the flags mean: secp, spsp and nssf.