linux-nvme/nvme-cli #408: nvme dsm command returns INVALID_OPCODE

erthalion, 2018-08-31:
Hi,

I'm trying to use a `dsm` command, but it returns INVALID_OPCODE error:

$ cat /testfile

keithbusch, 2018-08-31:
Please send the output of `nvme id-ctrl -H /dev/nvme0n1 | grep "Data Set Management"`

erthalion, 2018-08-31:
```
$ nvme id-ctrl -H /dev/nvme0n1 | grep "Data Set Management"
  [2:2] : 0x1   Data Set Management Supported
```

keithbusch, 2018-08-31:
Sounds like the controller your using has a firmware defect. The nvme-cli is just reporting what the controller returned to the requested command in this case, so if the controller claims to support DSM, then it shouldn't be an invalid opcode.

erthalion, 2018-09-02:
Looks like the very same command works for other instance types, so it's indeed AWS EC2 problem.

closed 2018-09-02 by erthalion