linux-nvme/nvme-cli #412: 4286 error when executing nvme format

sagcan, 2018-09-10:
Hi,

I recently was experiencing performance issues so I thought I'd try `nvme-cli`. Having read the docs I executed the command `sudo nvme format /dev/nvme0n1` without any success. I am getting the following error: 

> NVME Admin command error:ACCESS_DENIED: Access to the namespace and/or LBA range is denied due to lack of access rights(4286)

`# nvme id-ctrl /dev/nvme0n1` outputs the following model name LENSE20512GMSP34MEAT2TA which apparently is a NVMe device created by [Lenovo?](https://fccid.io/MSIP-REM-lnv-AM6671)
Anyways, is there any solutions to execute the command?

Thanks in advance :)

keithbusch, 2018-09-10:
Sounds like your device has a security feature that's locked the drive. Unfortunately security implementation is vendor specific (or at least not defined by NVMe), so nvme-cli doesn't have device specific unlocking capabilities.

sagcan, 2018-09-10:
Thanks for the reply! So I checked if there is any option in the BIOS to "unlock" the drive but I was unsuccessful. After doing some research I finally stumbled upon the [Thinkpad Drive Erase Utility](https://support.lenovo.com/us/en/downloads/ds019026) so I guess you're right. For anybody with the same issue, just check out the link! I'll test this today and report back. In any case I guess the issue is solved / closed?

sagcan, 2018-09-11:
So I erased all contents of my drive with the [Thinkpad Drive Erase Utility](https://support.lenovo.com/us/en/downloads/ds019026) and the performance is indeed better than before. Too bad `nvme-cli` couldn't be executed but the issue is solved.

closed 2018-09-11 by sagcan

pitrk, 2019-07-28:
If you have a Lenovo E590 with similar problem and Thinkpad Drive Erase utility does not support your laptop, go to BIOS and run Secure Wipe.

tomty89, 2021-05-07:
[Hope you don't mind me spreading the possible solution]
See https://github.com/linux-nvme/nvme-cli/issues/816#issuecomment-834586681

bughunter2, 2023-05-05:
As mentioned by tomty89, a possible solution may be to suspend/resume your computer.
If you're using a console in a Linux live or rescue environment, that boils down to:
`echo mem | sudo tee /sys/power/state`
Afterwards you might be able to use 'nvme format' successfully.